openshwprojects/OpenBK7231T_App #298: [Help & testing needed] realloc seems to be broken on T platform, but works on N

openshwprojects, 2022-10-17:
I have never used realloc on OBK before, but it seems like it's broken on T platform.
realloc is used cJSON_Print, and it hangs device where realloc is called.
As a workaround, I disabled all reallocs calls in cJSON on T platform:
https://github.com/openshwprojects/OpenBK7231T_App/commit/a88140c583a4193b3659da26c1d041b3fc9c1dcb
And here I added some test console commands:
https://github.com/openshwprojects/OpenBK7231T_App/blob/main/src/cmnds/cmd_test.c
**If you're reading this, can you check on your devices the following commands:**
- testMallocFree- should work on T and N
- testRealloc- should hang on T
- testJSON - should work on T and N
**and post here results?**

Extra points for person who can figure out why realloc is broken on T, especially considering that heap4.c from RTOS seems very similar on both T and N.

iprak, 2022-10-18:
I think I might have into something like this which is why I switched to `cJSON_PrintPreallocated`.

iprak, 2022-10-18:
Do you know where FIX_REALLOC_ISSUE might be getting set? https://github.com/openshwprojects/OpenBK7231T/blob/eaf4e1a606571b9f13d9c933da61670c712a6c8d/platforms/bk7231t/bk7231t_os/beken378/os/mem_arch.c#L30

openshwprojects, 2022-10-18:
@iprak , I don't know, but this is a good find. I didn't see it before. Maybe FIX_REALLOC_ISSUE is defined on N but not on T?

iprak, 2022-10-18:
That function for N seems to be similar but there are other difference in the file itself. But still it might be worth a try.

openshwprojects, 2022-10-23:
@iprak  os_realloc works on T, but realloc doesn't

openshwprojects, 2022-10-23:
https://github.com/openshwprojects/OpenBK7231T_App/commit/e77d66292c0cac21226f9591b9ceb8118e973b7e

solved? still, if you can, please check

closed 2022-10-23 by openshwprojects

iprak, 2022-10-23:
Confirmed: all 3 test commands pass on T.