openshwprojects/OpenBK7231T_App #299: New build errors/warnings

iprak, 2022-10-18:
@valeklubomir #293 introduced some build errors. @openshwprojects I am surprised that commit hook did not catch that.

1. The first param is incorrect
```
      hprintf128("<h2>Periodic Statistics</h2><h5>Consumption (during this period): ");
```
https://github.com/iprak/OpenBK7231T_App/blob/5a82465b828c08e588d19da1e0ef61ae32684ca0/src/driver/drv_bl_shared.c#L80

2. This line feels incorrect. It should be comparing, right?
```
    if (energyCounterStatsEnable = true)
```
https://github.com/iprak/OpenBK7231T_App/blob/5a82465b828c08e588d19da1e0ef61ae32684ca0/src/driver/drv_bl_shared.c#L375

3. This should use %ld?
```
sprintf(tmp, "<br>History Index: %d<br>JSON Stats: %s </h5>", energyCounterMinutesIndex,
				(energyCounterStatsJSONEnable == true) ? "enabled" : "disabled");
```

openshwprojects, 2022-10-18:
@iprak huh thanks, you are quick, there were indeed silly mistakes.
The hprintf128 is on my side, and the statsEnable = true must have been @valeklubomir , I already had a "wtf" moment few hours ago where the stats system was started despite me changing the initial value to false, and now I see why it was started! Thanks. 
Ref 1 and 2 - fixed, 3 - wait let me check docs

iprak, 2022-10-18:
Sounds good. I was just doing a final build check before pushing.

closed 2022-10-18 by iprak