openshwprojects/OpenBK7231T_App #301: Git conflict

valeklubomir, 2022-10-19:
file src/driver/drv_ir.cpp is not compiled 
Debug/obj/drv_ir.cpp: No such file or directory

clean erases this file and compilation if successful, but git status report deleted file.

Extension CPP somehow is not considered as source file.

btsimonh, 2022-10-19:
yep, working on it.
see https://github.com/openshwprojects/OpenBK7231T_App/pull/281

btsimonh, 2022-10-19:
p.s. for local T dev, ensure to pull latest T SDK.

valeklubomir, 2022-10-19:
I am working with N version. Don't have any T device to work with. 
Can you not rename it simple C?

openshwprojects, 2022-10-19:
IRRemote is a C++ library. Someone needs to backport those C++-enabling changes to N:
https://github.com/openshwprojects/OpenBK7231T/commits/master
@btsimonh  might do it later?

btsimonh, 2022-10-19:
looking at it quickly now.
@valeklubomir - I'll point you to a branch to test, may be a few minutes...

valeklubomir, 2022-10-19:
When I delete drv_ir.cpp N build works. 
No need to hurry. I keep in mind that not to commit deletion to git.

openshwprojects, 2022-10-19:
@valeklubomir  I am afraid that my next commit will break your compilation, unless you comment out IR line here:
[image]

valeklubomir, 2022-10-19:
Link it to T version only 
#if PLATFORM_BK7231T

btsimonh, 2022-10-19:
@valeklubomir -
please can you test the mods in the SDK here:
https://github.com/btsimonh/tuya-iotos-embeded-sdk-wifi-ble-bk7231n/tree/AddCPPSupport

valeklubomir, 2022-10-19:
tuya-iotos-embeded-sdk-wifi-ble-bk7231n/platforms/bk7231n/bk7231n_os/beken378/func/lwip_intf/lwip-2.1.3/port/ethernetif.c:140: undefined reference to `CFG_GetDeviceName'

openshwprojects, 2022-10-19:
@valeklubomir  don't you need to do the same as me before, the update-submodules?

btsimonh, 2022-10-19:
deleted from debug?  or deleted from source folder?
It could be that the stuff from the previous (bad) build in debug caused a problem...
I hate the way they have done that....

valeklubomir, 2022-10-19:
Deleted from source directory by clean command.

valeklubomir, 2022-10-19:
When I checkout fresh OpenBK7231T_App from git into app/OpenBK7231N_APP and execute script ./b_rebuild.sh 
It resulted with:
arm-none-eabi-gcc: error: Debug/obj/drv_ir.cpp: No such file or directory

src/driver/drv_ir.cpp is in the repositiory.

When I execute script: ./b_full.sh
which first call clean, then the build is successfully completed and binary is created.

after this src/driver/drv_ir.cpp is deleted and git status report it. With next commit file would be removed from repository.

I don't use make from App. It work only for N and T, other platforms fails, and it take time to build all of them.
At moment I work only with N versions. Do not have any other device for real testing then BK7231N

openshwprojects, 2022-10-19:
I pushed some fixes and warning removal (not all yet).

@valeklubomir , depending whether we can resolve N build, I might make IR driver T-only so you don't have troubles compiling..

At the current state of things, the local Windows PC build works for me for T platform, while building with cygwin.

btsimonh, 2022-10-19:
please can you check that application.mk definitely includes the line:
SRC_CPP += $(foreach dir, $(TY_SRC_DIRS), $(wildcard $(dir)/*.cpp)) 

remember, for local build, you probably have the SDK in use being outside of the OpenBeken folder:

[image]

NOT the SDKs inside the app folder?

valeklubomir, 2022-10-19:
No entry in any SDK
openshwprojects/OpenBK7231T
openshwprojects/OpenBK7231N
btsimonh/tuya-iotos-embeded-sdk-wifi-ble-bk7231n/tree/AddCPPSupport

btsimonh, 2022-10-19:
https://github.com/btsimonh/tuya-iotos-embeded-sdk-wifi-ble-bk7231n/blob/AddCPPSupport/platforms/bk7231n/bk7231n_os/application.mk

line 1106?

valeklubomir, 2022-10-19:
Needed to checkout diffrent branch. 
Now it works correctly

btsimonh, 2022-10-19:
we got there in the end :)  thankyou for your patience!

valeklubomir, 2022-10-19:
@btsimonh I ment your repos.
Original I receive 
Debug/obj/drv_main.o:(.data.g_drivers+0xc4): undefined reference to DRV_IR_Init' Debug/obj/drv_main.o:(.data.g_drivers+0xc8): undefined reference to DRV_IR_RunFrame'
collect2: error: ld returned 1 exit status

btsimonh, 2022-10-19:
I think this is WIP ref @openshwprojects - i.e. although the functions are in the drv_ir.cpp, I suspect he needs to add N platform to enable them for you too.  (look at #defines?).  Just shut down my OpenBeken dev laptop.... so can't check.

looking direct at github, the whole file has #if PLATFORM_BK7231T around it.

try:

#if PLATFORM_BK7231T || PLATFORM_BK7231N

that may work :).

btsimonh, 2022-10-19:
or 
#if PLATFORM_BEKEN

openshwprojects, 2022-10-19:
I was out for a 15 minutes or so, @valeklubomir  if that's a fix can you submit a pull request?

btsimonh, 2022-10-19:
"removed #if PLATFORM_BK7231T from drv_ir.cpp"
it probably needs 
#if PLATFORM_BEKEN
else other builds may break.

closed 2022-10-19 by valeklubomir