openshwprojects/OpenBK7231T_App #306: Tuya g9 bulb

bonuzzz, 2022-10-19:
It was tuya g9 bulb from aliexpress. Please, help me to recognize this board. I'm 99% sure it is openbeken board as tuya-cloudcutter works there. I want to dump firmware, but I can't recognize board. Pins depend on it
Thanks in advance

bonuzzz, 2022-10-19:
Seems it is bk7231n, but without soldered pins... Only now I notice this board doesn't have opened pins. To be honest, I have not idea how to flash it in this case

openshwprojects, 2022-10-20:
What is written on QFN chip? I can't see well. What is the chip name?

bonuzzz, 2022-10-20:
t34
2124k8wyn
x00leb

openshwprojects, 2022-10-21:
Where to buy this bulb? I could try to solder very small wires to pads, @bonuzzz 
@ferbulous  I think he doesn't have a profile, he just thinks that cloudcutter could help

bonuzzz, 2022-10-21:
@openshwprojects I bought it in aliexpress. Search "tuya g9 bulb". I don't want to recommend current shop, because it didn't want to return money for non working bulb (only cold light leds were working).
What about cloudcutter, I don't have working profile.

I attach in zip some photos without compression. I'm not sure it's possible to solder wires without unsoldering, but I'm not sure as well is it possible to power on pure chip without any resistors etc.
[tuya-g9.zip](https://github.com/openshwprojects/OpenBK7231T_App/files/9837751/tuya-g9.zip)

ferbulous, 2022-10-21:
@bonuzzz you might need share to the shops link because different shop might sell different chip for the g9

none of the profile worked with cloudcutter?

bonuzzz, 2022-10-21:
I bought them here: https://aliexpress.ru/store/911420118?spm=a2g2w.orderdetails.0.0.41eb4aa6ScEUcl
I know about awful rating. don't tell anything:) Unfortunately I didn't find them in normal shops.
I doubt about different revisions, because there are new bulbs in g9 form factor.
I think I could unsolder the chip myself, but I'm not sure will it work without other resistors

openshwprojects, 2022-10-21:
@bonuzzz  thank you for sending a zip file with photos, but if you want it to be featured on out teardowns list here:
https://openbekeniot.github.io/webapp/examples/devicesListDemo2.html
you need to post teardown here: https://www.elektroda.com/rtvforum/forum507.html
Post following things:
1. screenshot from where to buy
2. device name, model
3. packaging photos (if possible)
4. device photos from inside and outside (teardown)
5. obk config (if you managed to flash)

In a meantime, I will try to order this bulb, if it ships to Poland as well

@bonuzzz  do not 'unsolder the chip', you'd need to solder wires to chip, check T34 pinout on Tuya website and find where the programming interface is

bonuzzz, 2022-10-21:
@openshwprojects I don't have any ideas how to flash, because all the pins are hidden.

openshwprojects, 2022-10-21:
[image]
the only possible option is to get a very thin wire, and a microscope and try to solder directly to QFN legs...

ferbulous, 2022-10-21:
Another tool i would recommend is this pcb.[workstation](https://www.thingiverse.com/thing:3615910) 

Since it’s using needle , you could probe directly to the small pins and stick it with some blue tacs.

These pins are all exposed on each sides right like in this pic? Looks like gnd & vcc is on this side
@openshwprojects do you think the rx & tx pins are at the top

[image]
[image]

bonuzzz, 2022-10-21:
I lost my attempt. I unsoldered chip. Soldered the wires, but some of them did tear off and broke few pads on chip

bonuzzz, 2022-10-21:
I had very hard wires for it... On 2nd attempt I recovered all pins except cen and at this time I lost power pin. I orderd pogo pins. Next time will print holder for chip and use pogo.
Thanks all for your attention:)

closed 2022-10-21 by bonuzzz