openshwprojects/OpenBK7231T_App #310: Config IRRecv for Pin results in broken device, BK7231N

Handy-E, 2022-10-20:
Tried the new OpenBK7231T release 1.14.34 because I own a RGB-LED-stripe with WIFI and IR. Module is ([CBU](https://developer.tuya.com/en/docs/iot/cbu-module-datasheet?id=Ka07pykl5dk4u)) with BK7231N.
The IR-receiver seems connected to Pin1 of the Module. According to datasheet this should be pin P14.
Setting P14 to IRRecv in WebGUI results in broken Chip after Restart. No WebUI, no AP, no Wifi, nothing. The Module just gets very warm!
The only way I was able to fix the whole thing was by uploading (with hw-programmer) a version without IR-support, setting P14 to nothing and ota-upgrading again.
New try: Set P14 to IRRecv again => Device dead.

openshwprojects, 2022-10-20:
I havent tried N platform yet, but but.... @Handy-E , are you really sure about that flashing non-IR version? I'd assume that Safe Mode would still be working and the IR driver would not start in Safe Mode, so you can easily recover device without reconnecting the flashing wires...

Handy-E, 2022-10-20:
As I wrote, tried release 1.14.34, cannot find a newer one currently!
As "initial" version (non-IR) I used "OpenBK7231N_QIO_1.14.13.bin".
Prehaps I tried not hard enough for SafeMode ;-) How often do I have to reset? Readme says "If you power off and on device multiple times". What means "multiple" here ?
Thanks in advance! Good work, still!

btsimonh, 2022-10-20:
5 times worked for me today :).  then it makes 192.168.4.1 ....
May be good to get a serial log of boot and non-boot.

Handy-E, 2022-10-20:
> May be good to get a serial log of boot and non-boot.
Will try. TX2 and RX2, right? Baudrate?

btsimonh, 2022-10-20:
what's the temperature stuff about?  should not have that for an LED strip?  Odd...
The IR driver is scheduled to start after 5 seconds.  hence last log is at 4.
To me, it would seem that the N driver for bk_timer is somehow hooked up to some adc driver designed to read temperature.
Unless you want to dig in the N SDK, we should abandon on N until someone has worked it out....
possibly search for bk_timer and adc_trend?

Handy-E, 2022-10-20:
> what's the temperature stuff about? should not have that for an LED strip? Odd...

I did not configure anything to ADC, just IRRecv ;-)
> To me, it would seem that the N driver for bk_timer is somehow hooked up to some adc driver designed to read temperature.

Overtemperature-protection ???
> Unless you want to dig in the N SDK
possibly search for bk_timer and adc_trend?

Sorry, my knowledge to C\C++ is limited.

Handy-E, 2022-10-31:
> I havent tried N platform yet, but but.... @Handy-E , are you really sure about that flashing non-IR version? I'd assume that Safe Mode would still be working and the IR driver would not start in Safe Mode, so you can easily recover device without reconnecting the flashing wires...

@openshwprojects did you find something here. I looked at the SDK but did not find anything useful. But, as I wrote, my knowledge here is limited, sorry. So far, the timer seems somehow bundled to adc. But the original firmware also used IR, so it has to be possible in some way.
So far, still, I can't use IR and my remote. :-(

Handy-E, 2022-11-01:
So cool! Newest version fixed it. IR seems working now! Thanks to all devs!!!

closed 2022-11-01 by Handy-E

btsimonh, 2022-11-01:
yep, was my bad, re-initialising the system timers when they were already initialised.  I now have a N based IR blaster, which I used to diagnose and fix the issue.  In coming weeks will put it to use controlling 13 IR controlled LED lights which I have in one room :).