openshwprojects/OpenBK7231T_App #311: EDM-01AA-EU dimmer blinking

leoshusar, 2022-10-20:
Hi! I've just flashed my EDM-01AA-EU TuyaMCU dimmer and used [this](https://templates.blakadder.com/unsupported/eardatek_EDM-01AA-EU.html) Tasmota template.

It works fine, but when I turn it on, the whole brightness bar blinks like 2 or 3 times per second, I can even set brightness physically while it's blinking, so it's not blocking anything.
It shouldn't be pairing mode, because that's only single bottom LED.

I've found [this](https://www.elektroda.com/rtvforum/viewtopic.php?p=19843655#19843655) post where you already have this exact dimmer - is it working for you fine? Did you make anything else to make it working?
My autoexec:
```
startDriver TuyaMCU
setChannelType 1 toggle
setChannelType 2 dimmer
tuyaMcu_setBaudRate 115200
tuyaMcu_setDimmerRange 1 1000
linkTuyaMCUOutputToChannel 1 bool 1
linkTuyaMCUOutputToChannel 2 val 2
```

I also have the original FW dump if you want, but I have my WiFi password in there.

And here is somehow cleaned piece of TuyaMCU log since boot:
```
Info:TuyaMCU:TUYAMCU received: 55 AA 03 02 00 00 04 
Info:TuyaMCU:TuyaMCU_ProcessIncoming[ver=3]: processing command 2 (MCUconf) with 7 bytes
Info:TuyaMCU:TuyaMCU_ProcessIncoming: TUYA_CMD_MCU_CONF had wrong data lenght?

Info:TuyaMCU:TUYAMCU received: 55 AA 03 07 00 05 01 01 00 01 00 11 
Info:TuyaMCU:TuyaMCU_ProcessIncoming[ver=3]: processing command 7 (State) with 12 bytes
Info:TuyaMCU:TuyaMCU_ParseStateMessage: processing dpId 1, dataType 1-DP_TYPE_BOOL and 1 data bytes
Info:TuyaMCU:TuyaMCU_ParseStateMessage: raw data 1 byte: 
Info:GEN:No change in channel 1 (still set to 0) - ignoring

Info:TuyaMCU:TUYAMCU received: 55 AA 03 07 00 08 02 02 00 04 00 00 01 F4 0E 
Info:TuyaMCU:TuyaMCU_ProcessIncoming[ver=3]: processing command 7 (State) with 15 bytes
Info:TuyaMCU:TuyaMCU_ParseStateMessage: processing dpId 2, dataType 2-DP_TYPE_VALUE and 4 data bytes
Info:TuyaMCU:TuyaMCU_ParseStateMessage: raw data 4 int: 500
Info:GEN:CHANNEL_Set channel 2 has changed to 49 (flags 0)

Info:TuyaMCU:TUYAMCU received: 55 AA 03 07 00 08 03 02 00 04 00 00 00 0A 24 
Info:TuyaMCU:TuyaMCU_ProcessIncoming[ver=3]: processing command 7 (State) with 15 bytes
Info:TuyaMCU:TuyaMCU_ParseStateMessage: processing dpId 3, dataType 2-DP_TYPE_VALUE and 4 data bytes
Info:TuyaMCU:TuyaMCU_ParseStateMessage: raw data 4 int: 10

Info:TuyaMCU:TUYAMCU received: 55 AA 03 00 00 01 01 04 
Info:TuyaMCU:TuyaMCU_ProcessIncoming[ver=3]: processing command 0 (Hearbeat) with 8 bytes
Info:TuyaMCU:TUYAMCU received: 55 AA 03 03 00 00 05 
Info:TuyaMCU:TuyaMCU_ProcessIncoming[ver=3]: processing command 3 (WiFiState) with 7 bytes
```

leoshusar, 2022-10-20:
It... fixed itself while messing with MQTT.
Then I tried to power cycle it, it was blinking again but after toggle off and on it stopped.

closed 2022-10-20 by leoshusar

openshwprojects, 2022-10-21:
Oh ok, just tell me if you need anything else, I will do my best to help @leoshusar