openshwprojects/OpenBK7231T_App #314: Exec of a file kills T on next Wifi stat display

btsimonh, 2022-10-21:
I created a file IR.txt containing 

```
addEventHandler2 IR_NEC 0x0 0x48 add_dimmer 10
addEventHandler2 IR_NEC 0x0 0x4C add_dimmer -10
```

on issuing `exec IR.txt`, the device add the commands, but dies at the next wifi poll.
I did not use the IR commands.... just added them through exec of an FS file.

Additional: tried commands on their own, not using a file.  One time out of 4 adding a command killed the device.

log:
```
do td cur_t:288--last:idx:13,t:280 -- new:idx:14,t:292
--0xc:08, shift_b:0, shift_g:0, X:2
                                   Info:MAIN:Time 9, free 98208, MQTT 0(0), bWifi 1, secondsWithNoPing 0, socks 2/38
sta: 1, softap: 0, b/g/n
sta:rssi=-54,ssid=MYSSID,bssid=00:37:b7:30:e0:0e ,channel=11,cipher_type:CCMP
Info:MAIN:Time 10, free 98208, MQTT 0(0), bWifi 1, secondsWithNoPing 0, socks 2/38
Info:GEN:dhcp=0 ip=192.168.1.176 gate=192.168.1.1 mask=255.255.255.0 mac=7c:f6:66:0b:3f:e8
Info:MAIN:Time 11, free 98208, MQTT 0(0), bWifi 1, secondsWithNoPing 0, socks 2/38
Info:MAIN:Time 12, free 98208, MQTT 0(0), bWifi 1, secondsWithNoPing 0, socks 2/38
Info:MAIN:Time 13, free 98208, MQTT 0(0), bWifi 1, secondsWithNoPing 0, socks 2/38
Info:MAIN:Time 14, free 98208, MQTT 0(0), bWifi 1, secondsWithNoPing 0, socks 2/38
Info:MAIN:Time 15, free 98208, MQTT 0(0), bWifi 1, secondsWithNoPing 0, socks 2/38
Info:MQTT:mqtt_userName admin
mqtt_pass password
mqtt_clientID obk660B3FE8
mqtt_host raspberrypi:1883
Info:MAIN:Time 16, free 98208, MQTT 0(1), bWifi 1, secondsWithNoPing 0, socks 2/38
Info:MQTT:mqtt_connection_cb: Successfully connected
Info:MQTT:mqtt_subscribed to obk660B3FE8/+/set
Info:MQTT:mqtt_subscribed to cmnd/obk660B3FE8/+
Info:MQTT:MQTT client "obk660B3FE8" request cb: err 0
Info:MQTT:MQTT client "obk660B3FE8" request cb: err 0
Info:EVENT:CMD_AddEventHandler: will expect two args.
Info:EVENT:CMD_AddEventHandler: arg2 = 72.
Info:EVENT:CMD_AddEventHandler: arg1 = 0.
Info:EVENT:CMD_AddEventHandler: added IR_NEC with cmd add_dimmer 10
Info:EVENT:CMD_AddEventHandler: will expect two args.
Info:EVENT:CMD_AddEventHandler: arg2 = 76.
Info:EVENT:CMD_AddEventHandler: arg1 = 0.
Info:EVENT:CMD_AddEventHandler: added IR_NEC with cmd add_dimmer -10
Info:MAIN:Time 17, free 98088, MQTT 1(1), bWifi 1, secondsWithNoPing 0, socks 2/38
Info:MAIN:Time 18, free 98088, MQTT 1(1), bWifi 1, secondsWithNoPing 0, socks 2/38
Info:MAIN:Time 19, free 98088, MQTT 1(1), bWifi 1, secondsWithNoPing 0, socks 2/38
sta: 1, softap: 0, b/g/n
```

openshwprojects, 2022-10-21:
@btsimonh  can confirm. It happens for me if I use exec with such file as well:
[image]
when I apply those commands by hand, I never get any crashes. I just got first time in my life crash on this device, thanks to the method with exec...

openshwprojects, 2022-10-23:
I think we both nwo can agree it was solved by increasing the stack size of HTTP client.

closed 2022-10-23 by openshwprojects