openshwprojects/OpenBK7231T_App #315: Power Calibration not saved

daniel-hofer, 2022-10-21:
First - Thank you for the great Project!

After Flashing Tasmota to my Aubess Plug (new Chip) https://templates.blakadder.com/aubess_16A_plug.html
i tried to calibrate the power and voltage with VoltageSet 150.
But every power cycle or reboot sets it back to original config.
Values like the wifi or mqtt settings are saved.
SaveConfig 1 is not a known command.
Can you help me yout?
Thanks a lot,
regards Daniel

openshwprojects, 2022-10-21:
Hello @daniel-hofer . The current assumption is that user puts the commands for setting IRef in the short startup command manually, as said here:
https://www.elektroda.com/rtvforum/topic3887748.html
but since you're asking, it might really be a good idea to just save them...

Wait few hours, I will try to get some free time and do a simple autosave of calibrated values, so users don't have to do it by hand.

daniel-hofer, 2022-10-21:
Thank you for the fast answer and information to fix the problem. 
I'll try the startup command and look forward to your changes.

Great work!

openshwprojects, 2022-10-21:
@daniel-hofer  I pushed autosave for BL0937, BL0942 and CSE.

**Can you check if it works - now really, please? It was not tested so it would be good if someone can confirm if it autosave correctly values.**

valeklubomir, 2022-10-21:
@openshwprojects From my side saving calibration seems to be working. Testing more.
Today I received 3x BK7213N(CB2S) devices 1x BK7231T (WB3S) 2x ESP8266 devices.         :-)

openshwprojects, 2022-10-21:
@valeklubomir it would be great if you look into the T LWIP stability. Also, if you have new devices, you can make a photos and create some teardowns for our database:
https://openbekeniot.github.io/webapp/examples/devicesListDemo2.html
I want to get to the 100 devices soon. It's 94 so far.
post them on https://www.elektroda.com/rtvforum/forum507.html and I will add them to online database.

In a meantime, I will extend the IR functionality and scripting. I am also working on third and fourth OpenBeken youtube videos.

valeklubomir, 2022-10-21:
@openshwprojects so far each device which I tried, was already in library or similar device.
Once I have all devices completed, I will try to summarize it. 
I already imported LWIP to my local fork of T device. At least building works, not yet tested. 
New T Device (touch switch) is in my queue. 
I am still trying to figure out issues with freezing. BL0937 seems to be more sensitive than BL0942. 
Found few minor issues. Hoping if enough issues are fixed, it could stabilize FW for real day-by-day use.

openshwprojects, 2022-10-21:
@valeklubomir  what are those issues?

valeklubomir, 2022-10-21:
I think:
- logging large blocks or fast occuring events which could overflow serial/UDP logging buffers may cause memory leaks. My device froze mostly after JSON stat print when >1024 byte should be printed. Trying with disabled JSON as comparision.
- DRV HTTP plugin after some time displays rubbish when printing driver list, pointers to DRV names got corrupted. Printed random text string from memory. Searching......
- BL09XX HTTP plugin at least once broke front page. Probably htprint128 overrun. htprint128 can handle only 256 bytes per call, if somehow larger block is posted, it is now truncated. which can cause issues later for browser. There was pull request witn replacement of vsprintf with vsnprintf. Since heap should be omitted could not impement dynamic detection.

openshwprojects, 2022-10-21:
@valeklubomir  I've found one issue: 
https://github.com/openshwprojects/OpenBK7231T_App/commit/7f0cf7ad7d713d57a39350a4371a9bc06bd7a216

openshwprojects, 2022-10-21:
_BL09XX HTTP plugin at least once broke front page. Probably htprint128 overrun. htprint128 can handle only 256 bytes per call, if somehow larger block is posted, it is now truncated. which can cause issues later for browser. I already created pull request with replacement of vsprintf with vsnprintf. Since heap should be omitted could not impement dynamic detection. And it is already merged. But occurence os larger string must be still corrected._

that one should be easy, just split strings into more calls..

openshwprojects, 2022-10-21:
_Button active level set to 1, because **_n input are inverted. And active level must be 1. This was generating a lot of logginng messages showing button long press hold._

Really? The _n should be "transparent" for all stuff except this function:
[image]

The Button_n is a negated Button where instead of having a pull up resistor and button to ground one has a pull down resistor and button to VDD

valeklubomir, 2022-10-21:
BL0937 uses interrupt calls for measurring CF/CF1 signals. Even when simple functions used. 
Don't know how it behaves under BK7231 since do datasheet or manual is available, but I know that in mutlithreading applications interrupts should be extra carefully designed when dealing with threads. BK7231 should be ARM968e-s core. 
This need some study also.

Splitting strings and using sprintf rather htprint128 arguments.

valeklubomir, 2022-10-21:
Yes button_n is inversed, but after "transparency" active level remained 0 instead of 1.

[image]

openshwprojects, 2022-10-21:
_sprintf rather htprint128 arguments._

sprintf also takes a buffer as an argument, so buffer overrun still can happen, so I don't see how it could help? I'd just split HTML construction into more htprint128 calls and that's all

@valeklubomir  I might be wrong, but I think that active level should be the same for both _n button and simple button, because we do inversion BEFORE comparing with active level:
[image]
The button code 'doesn't know' whether it is an inverted (button_n) or a simple (button) button. Inversion is down one level below.
So the active level HAS TO remain 0 all the time, because the "swap" occurs earlier

openshwprojects, 2022-10-21:
@valeklubomir I think we might be taking past each other. For me, the "button_n" is a button with a pull down resistor and a button between input pin and a VDD. For you, it seems that a button_n is a button with a pull up resistor (like a button) and a button between GND and input pin, but which is normally closed all the time and just opens when a users does a press...?

valeklubomir, 2022-10-21:
Usually negated input is designed having pull-up and switched to GND for active level.

valeklubomir, 2022-10-21:
In electronics inputs marked '_n' are ment with inactive level H and active level L.

valeklubomir, 2022-10-21:
It is similar when you do negation of bool: "!(bool)"

openshwprojects, 2022-10-21:
@valeklubomir  there is another change I want to do (or have to, because right now it might cause problems).
I am worried about having so many threads. Right now, we have:
- quick tick thread (5ms or so), this is used for  buttons and for drivers quick tick
- every second timer
- HTTP threads
I am thinking about merging quick tick with 'every second timer' so we don't need mutexes in drv_main.c so often
Another think to do would be to don't use HTTP threads at all, just do a blocking sockets, then we would have on main thread (and HTTP clients could be handled in HTTP) but I remember that I tried that (used non-blocking sockets) and it failed - it was just freezing the device.

What do you think about merging 'quick tick' with 'on every second', at least on Beken platform, not now, but in general?

valeklubomir, 2022-10-21:
Merging timers could be easy. Keep 5ms and replace 1s with calling from 5ms with ratio 1:200. 
For example TuyaMCU run frame would benefit for calling with higher frequency. Example: 200 - 300 ms run.
Touch shields generate messages every 200/300 ms when you press and hold. And could react faster.

daniel-hofer, 2022-10-22:
> @daniel-hofer I pushed autosave for BL0937, BL0942 and CSE.
> 
> **Can you check if it works - now really, please? It was not tested so it would be good if someone can confirm if it autosave correctly values.**

Thank you very much for the fast fix! I can confirm, that the commnads VoltageSet, CurrentSet and PowerSet are saved now after reboot :-) My device is an Aubess Smart Plug Chipset BK7231N (CB2S - module). I'm so glad having this device now in my homeassistant dashboard! Tip is on the way!

openshwprojects, 2022-10-23:
@daniel-hofer  thank you, I'm glad I helped. If you have a device that is not known to us here:
https://openbekeniot.github.io/webapp/examples/devicesListDemo2.html
you can also help us and submit a teardown here:
https://www.elektroda.com/rtvforum/forum507.html
we really want to get up to 100 device teartodnws soon!

btsimonh, 2022-10-24:
@valeklubomir - "I already created pull request with replacement of vsprintf with vsnprintf."

can you point me to this?
I'm currently working on random crashes - mainly increased by lots of bk_printf, which uses vsnprintf, and I suspect that's killing it.

openshwprojects, 2022-10-24:
@btsimonh  https://github.com/openshwprojects/OpenBK7231T_App/commit/a3d8542173d154f4c6c600f8d909f6702b778f6d

valeklubomir, 2022-10-24:
vsnprintf should be safer version of vsprintf because extra parameter is used to limit max length of string. I expect that it should avoid writing over length of string memory.

openshwprojects, 2022-10-24:
@valeklubomir  if I understand correctly, @btsimonh  concern is not that is a buffer-size-safe version, he suspects that in general sprintf on RTOS platform we have is not thread safe... 
https://www.freertos.org/FreeRTOS_Support_Forum_Archive/April_2018/freertos_How_to_make_printf_sprintf_strtod_thread_safe_44d2ceb8j.html

openshwprojects, 2023-02-22:
Long time has passed and I think we know the calibration settings are now saved automatically. I can close this.