openshwprojects/OpenBK7231T_App #316: sum daily power meter values - big deal to implement?

mariusbach, 2022-10-21:
Hi, 
just asking if it's a big deal to implement daily Wh consumption counters? You could save the last value of the day as "yesterday's consumption" and display/publish the current count of Wh as displayed.

My goal is to place power metering plugs at the TV-hub, my working desk, the networking rack and so on and see a trend of daily values. 

Thanks for checking. I'd help with coding but I really have not much clue.

valeklubomir, 2022-10-22:
Device like this do only report power output.

Recently we added Total Consumption calculation and Periodic calculations. 
Periodic calculation could be configured to display consumption per last 24H.

I will investigate if statictics configured per last 24H is possible to use this way.
if device can be synced with NTP, so then at 23:59:59 could be stored as consumption 
over last day.

So far firmware does not storing consumption pernamently. After reset values are lost.
Device does not contains EEPROM memory, only FLASH memory, but this has limitations.
FLASH memories used to stored firmware do not have qualities to do many writes,
too many write operations could destroy device memory.

Calculating consumption inside the device can deliver higher precision, 
SmartLife/Tuya or similar cloud based system calculate this consumption from power  output
measured by device, but since value sent via MQTT, are not sent at very precise 
timing and when lost conenction calculation becomes even more unreliable.

This tasks, like daily/weekly/monthly statistics are ususally processed by higher system 
where tracking of collected consumption values over long period of time.

mariusbach, 2022-10-22:
Thanks for looking into this. Indeed, a higher system could do all the calculations (actually the total is already implemented so I can use it to check at midnight what the delta is), but a daily value in the firmware would save every one from programming his own routine on a higher system, also it wouldn't depend on the availability of the higher system. tasmota does show the daily accumulation and yesterday's total which I just feed into a database to have the value.