openshwprojects/OpenBK7231T_App #317: How to retain current status of device after HA restarts?

ferbulous, 2022-10-21:
Hi, I've enabled these settings in general

[image]

But when I restart HA, it doesn't show the current status and I still have to toggle it first to (either from the device or from HA dashboard) to have it display the correct status

[image]

Is there other settings I need to enable as well?

openshwprojects, 2022-10-22:
Hello @ferbulous , I will check this now. By the way, thanks for next teardown on Elektroda. We appreciate it. 

Let's test that on my device:
[image]

Hmm:
[image]
That's strange, I did a HA restart but the device didn't get MQTT 0 even for a second..

Info:MAIN:Time 333, free 85088, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 334, free 79536, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/38
Info:MAIN:Time 335, free 85088, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 336, free 85088, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 337, free 85088, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 338, free 85088, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 339, free 85088, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 340, free 85088, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38

I used this button:
[image]

So HA didn't have device state after restart and had to wait for "publish self state every minute".

That's strange.

Anyone of devs can confirm? @valeklubomir  @btsimonh 

Ok, let's do test 2 - turn off HA VM:
[image]

device still has MQTT 1:
[image]
Gotten time out:
[image]

MQTT still is 1, huh....
[image]

Info:MAIN:Time 591, free 85088, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 592, free 85088, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 593, free 85088, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 594, free 85088, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 595, free 85088, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 596, free 85088, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 597, free 85088, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 598, free 85088, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MQTT:Publishing val 1 to obk8D38570E/led_enableAll/get retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -9 result 0
Info:MAIN:Time 599, free 85088, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 600, free 79536, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/38
Info:GEN:dhcp=0 ip=192.168.0.117 gate=192.168.0.1 mask=255.255.255.0 mac=38:1f:8d:38:57:0e 
Info:MAIN:Time 601, free 85088, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MQTT:mqtt_connection_cb: Disconnected, reason: 257(Timeout)
Info:MAIN:Time 602, free 85088, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 603, free 85088, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 604, free 79536, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 3/38
Info:MAIN:Time 605, free 85088, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 606, free 85088, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 2/38

It detected disconnect after a minute or so....

Let me start HA now.
[image]
And device got 'reconnect' , by the way HA overwrote my device state (enable) but that's not the issue now..
[image]

**My current conclusion is that if you do a quick restart of Home Assistant, for some reason, the MQTT doesn't discover the connection drop, which is for me surprising because TCP should know that socket was closed... and since MQTT doesn't discover the connection loss, it does not broadcast the self state "on connect" and so the next time HA gets update is after a minute when a "broadcast self state every minute" runs.**
Thoughts, @valeklubomir  ?

@ferbulous can you post the full log from OBK , starting at the moment you restart HA and ending when HA is online? I would like to see if you also have MQTT 1 all the time.

openshwprojects, 2022-10-22:
The TCP connection of MQTT should know that it was dropped during restart and for some reason it doesn't seem to know. Maybe decreasing the allowed time out could help.

ferbulous, 2022-10-22:
@openshwprojects 
Hi, here's a recording of the logs when HA is restarted

https://www.youtube.com/watch?v=n1yAYNctIDU

> The TCP connection of MQTT should know that it was dropped during restart and for some reason it doesn't seem to know. Maybe decreasing the allowed time out could help.

How should I adjust that in the settings?

openshwprojects, 2022-10-22:
@ferbulous this is only available in code....

but there might be another approach - have you tried enabling 'retain' in flags?

ferbulous, 2022-10-22:
> @ferbulous this is only available in code....
> 
> but there might be another approach - have you tried enabling 'retain' in flags?

Thanks! That fixed it

closed 2022-10-22 by ferbulous

openshwprojects, 2022-10-22:
@ferbulous  this, however, does not explain why HA restart does not send a TCP close to sockets.... but I'm happy that I helped.