openshwprojects/OpenBK7231T_App #320: Zmai-90 support with new Tuya module - CBU

Pgeup, 2022-10-22:
Good day. Does OpenBeken support the new Tuya - CBU module in Zmai-90?

If yes, please tell me what is the pinout? When flashing the CBU module with the OpenBK7231N_QIO_1.14.42 file, an open access point is created, but there is no data in the interface and on the LCD screen of the counter

valeklubomir, 2022-10-22:
[https://www.creationfactory.co/2019/12/reverse-engineering-zmai-90-din-rail.html](url)
This version runs on TYW3S module (ESP8266 based), but the pin assignmet could help
Regarding the emetering module (here V9821), this is not yet supported by the OpenBK.
If you post some pictures could help.

openshwprojects, 2022-10-22:
@Pgeup  I have that Zmai-90. I will look into that.

Pgeup, 2022-10-22:
I am attaching a photo of the CBU module. [If](url) necessary, I can attach a dump of the original tuya firmware.

https://photos.app.goo.gl/YFpRp6hJsRhWY1KP9

openshwprojects, 2022-10-23:
@Pgeup  thanks, but it would be great if you made some more detailed photos, inside and from the outside, maybe packaging photos or screenshots from shop and post them here:
https://www.elektroda.com/rtvforum/forum507.html
so we can add them to our online database:
https://openbekeniot.github.io/webapp/examples/devicesListDemo2.html

This is especially useful because I have different zmai - I have the WB3S-based one.

efaden, 2022-10-25:
I have a Moe's 4 way smart switch with the same module.
[image]
[image]