openshwprojects/OpenBK7231T_App #321: Device Group issues when enabled

ferbulous, 2022-10-22:
Tested with the following devices:
a. Tasmota E27 Led
b. Openbkt E27 Led (N device). I'm not sure if it's device related since I only have N now.

Enabled:
1) Toggle on/off works but not for brightness & color change (done directly on the openbk ui). On homeassistant dashboard, i could toggle from the tasmota device but if I did the same for openbk issue (3) occurs
2) Sometimes repeated group command occurs (e.g keeps toggling on/off for 2-4 secs before it stops)
3) Openbk crashes when sending mqtt payload from homeassistant. Had to power cycle again when this happens

More info in this video:

https://youtu.be/ju0Mn1QNq1I

openshwprojects, 2022-10-23:
Today I have increased some stack sizes, can you recheck?

Regarding MQTT crash - @valeklubomir , ideas?

Regarding "blinking" - it might be because current DGR does not ack, but when I tested it, it was stable. Might improve it soon.

ferbulous, 2022-10-23:
Hi, 
Issue 2 is  still there
Issue 3 - I can send mqtt payload the first time, openbk does receive it and then it crashes

https://youtu.be/btTxuRLcuPw

Some of the logs during testing:
``
Info:DGR:DGR_Parse: [192.168.1.185] seq 13824, flags 2049
Info:MQTT:MQTT deduper sent 55, culled 260
Info:MAIN:Time 143, free 80496, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:DGR:DGR_Parse: [192.168.1.185] seq 3584, flags 0
Info:DGR:DGR_Parse: [192.168.1.109] seq 18176, flags 1
Info:DGR:DGR_Parse: [192.168.1.109] seq 18432, flags 1
Info:DGR:DGR_Parse: [192.168.1.185] seq 18176, flags 2049
Info:DGR:DGR_Parse: [192.168.1.185] seq 18432, flags 2049
Info:DGR:DGR_Parse: [192.168.1.109] seq 18944, flags 1
Info:DGR:DGR_Parse: [192.168.1.185] seq 18944, flags 2049
Info:DGR:DGR_Parse: [192.168.1.109] seq 19200, flags 1
Info:DGR:DGR_Parse: [192.168.1.185] seq 19200, flags 2049
Info:DGR:DGR_Parse: [192.168.1.185] seq 3840, flags 0
Info:MQTT:MQTT deduper sent 58, culled 279
Info:MAIN:Time 152, free 83480, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:DGR:DGR_Parse: [192.168.1.109] seq 19456, flags 1
Info:DGR:DGR_Parse: [192.168.1.109] seq 19712, flags 1
Info:DGR:DGR_Parse: [192.168.1.185] seq 19456, flags 2049
Info:DGR:DGR_Parse: [192.168.1.185] seq 19712, flags 2049
Info:DGR:DGR_Parse: [192.168.1.109] seq 20480, flags 1
Info:DGR:DGR_Parse: [192.168.1.109] seq 20736, flags 1
Info:DGR:DGR_Parse: [192.168.1.185] seq 20480, flags 2049
Info:DGR:DGR_Parse: [192.168.1.185] seq 20736, flags 2049
Info:DGR:DGR_Parse: [192.168.1.109] seq 20992, flags 1
Info:DGR:DGR_Parse: [192.168.1.109] seq 21248, flags 1
Info:DGR:DGR_Parse: [192.168.1.185] seq 20992, flags 2049
Info:DGR:DGR_Parse: [192.168.1.185] seq 21248, flags 2049
Info:DGR:DGR_Parse: [192.168.1.185] seq 3840, flags 0
Info:MQTT:MQTT deduper sent 58, culled 288
Info:MAIN:Time 153, free 83480, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:DGR:DGR_Parse: [192.168.1.109] seq 22272, flags 1
Info:DGR:DGR_Parse: [192.168.1.109] seq 22528, flags 1
Info:DGR:DGR_Parse: [192.168.1.109] seq 22784, flags 1
Info:DGR:DGR_Parse: [192.168.1.185] seq 22272, flags 2049
Info:DGR:DGR_Parse: [192.168.1.185] seq 22528, flags 2049
Info:DGR:DGR_Parse: [192.168.1.185] seq 22784, flags 2049
Info:DGR:DGR_Parse: [192.168.1.109] seq 23040, flags 1
Info:DGR:DGR_Parse: [192.168.1.109] seq 23296, flags 1
Info:DGR:DGR_Parse: [192.168.1.109] seq 23552, flags 1
Info:DGR:DGR_Parse: [192.168.1.185] seq 23040, flags 2049
Info:DGR:DGR_Parse: [192.168.1.185] seq 23296, flags 2049
Info:DGR:DGR_Parse: [192.168.1.185] seq 23552, flags 2049
Info:MQTT:MQTT deduper sent 58, culled 297
Info:MAIN:Time 154, free 83480, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:DGR:DGR_Parse: [192.168.1.109] seq 23808, flags 1
Info:DGR:DGR_Parse: [192.168.1.109] seq 24064, flags 1
Info:DGR:DGR_Parse: [192.168.1.109] seq 24320, flags 1
Info:DGR:DGR_Parse: [192.168.1.185] seq 23808, flags 2049
Info:DGR:DGR_Parse: [192.168.1.185] seq 24064, flags 2049
Info:DGR:DGR_Parse: [192.168.1.185] seq 24320, flags 2049
Info:DGR:DGR_Parse: [192.168.1.185] seq 3840, flags 0
Info:DGR:DGR_Parse: [192.168.1.109] seq 25600, flags 1
Info:DGR:DGR_Parse: [192.168.1.109] seq 25856, flags 1
Info:DGR:DGR_Parse: [192.168.1.109] seq 26112, flags 1
Info:DGR:DGR_Parse: [192.168.1.185] seq 25600, flags 2049
Info:DGR:DGR_Parse: [192.168.1.185] seq 25856, flags 2049
Info:DGR:DGR_Parse: [192.168.1.185] seq 26112, flags 2049
Info:MQTT:MQTT deduper sent 58, culled 307
Info:MAIN:Time 155, free 81352, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:DGR:DGR_Parse: [192.168.1.109] seq 26624, flags 1
Info:MQTT:Publishing val 0 to smartrul_bulb1/led_enableAll/get retain=1
Info:MQTT:MQTT deduper sent 59, culled 307
[... 282 lines cut ...]
Info:MAIN:Time 24, free 83944, MQTT 1(1), bWifi 1, secondsWithNoPing 0, socks 4/38
Info:MQTT:Publishing val Build on Oct 23 2022 06:52:52 version 1.14.48 to smartrul_bulb1/build retain=1
Info:MQTT:[g_bPublishAllStatesNow] item -12 result 0
Info:MQTT:MQTT deduper sent 0, culled 0
Info:MAIN:Time 27, free 83944, MQTT 1(1), bWifi 1, secondsWithNoPing 0, socks 4/38
Info:MQTT:MQTT deduper sent 0, culled 0
Info:MAIN:Time 31, free 83944, MQTT 1(1), bWifi 1, secondsWithNoPing 0, socks 4/38
Info:MAIN:Boot complete time reached (30 seconds)
Info:CFG:####### Set Boot Complete #######
Info:MQTT:MQTT deduper sent 0, culled 0
Info:MAIN:Time 32, free 83944, MQTT 1(1), bWifi 1, secondsWithNoPing 0, socks 4/38
Info:MQTT:Publishing val 100 to smartrul_bulb1/led_dimmer/get retain=1
Info:MQTT:[g_bPublishAllStatesNow] item -7 result 0
Info:MQTT:MQTT deduper sent 3, culled 0
Info:MAIN:Time 39, free 71816, MQTT 1(1), bWifi 1, secondsWithNoPing 0, socks 5/38
Info:MQTT:MQTT deduper sent 3, culled 0
Info:MAIN:Time 43, free 71816, MQTT 1(1), bWifi 1, secondsWithNoPing 0, socks 5/38
Info:MQTT:MQTT deduper sent 3, culled 0
Info:MAIN:Time 47, free 83512, MQTT 1(1), bWifi 1, secondsWithNoPing 0, socks 4/38
Info:MQTT:MQTT deduper sent 3, culled 0
Info:MAIN:Time 50, free 81384, MQTT 1(1), bWifi 1, secondsWithNoPing 0, socks 4/38
Info:GEN:dhcp=0 ip=192.168.1.109 gate=192.168.1.1 mask=255.255.255.0 mac=d8:1f:12:aa:1e:0b 
Info:MQTT:Publishing val 81176 to smartrul_bulb1/freeheap retain=1
Info:MQTT:[g_bPublishAllStatesNow] item -2 result 0
```

ferbulous, 2022-10-30:
Hi, thanks for the fixes with latest version 1.14.91
Changing brightness, and colors are working fine
The flickering seems to be gone

I can’t seem to figure out why openbeken still crashes if mqtt payload is sent from homeassistant/mqtt client with device group enabled.
I might need to connect to tx2 pin to get some log output.

btsimonh, 2022-11-04:
@ferbulous - i'll close this one for now, as original issue is resolved.
Do ping if the MQTT crashes are not fixed by  .123

closed 2022-11-04 by btsimonh

ferbulous, 2022-11-05:
@btsimonh hi, unfortunately the MQTT crash occurs remains when i tested on .124

Any useful logs i could share or some other settings i need to enable in openbk to prevent this?

btsimonh, 2022-11-05:
yeah, serial logging would be good at the point of the crash - especially if you ran 'logdelay 50' beforehand (this add a 50ms delay after each call to log, so the log can get out before the crash!!!).

but first, can you capture the actual MQTT (topic and data) which causes the crash - we can then test that directly by sending from Node Red, and see if it can be reproduced.

ferbulous, 2022-11-08:
> but first, can you capture the actual MQTT (topic and data) which causes the crash - we can then test that directly by sending from Node Red, and see if it can be reproduced.

I tried mqttexplorer & nodered, and tested this on/off payload when it crashes openbk
Device: led bulb

`cmnd/gu10_openbk_2/led_enableAll`

I tried sending brightness/temperature but I probably didn't get it right.
However if I change the brightness/temperature/color from ha dashboard it does crash openbk as well

[image]