openshwprojects/OpenBK7231T_App #322: MQTT Aubess Smart Plug no

daniel-hofer, 2022-10-22:
The Values for Power, Voltage, Current have no Unit. Is it possible to ad the Unit in the mqtt-payload?
Here The excample of my Aubess Plug - am I right, that the line "unit_of_measurement: V" would do the trick?
device:
  identifiers:
    - Aubess_Plug_Tasmota_01
  name: Aubess Plug Tasmota 01
  sw_version: 1.14.34
  manufacturer: Beken Corporation
  model: BK7231N
  configuration_url: http://192.168.1.99/index
name: Aubess Plug Tasmota 01 current
availability_topic: obk1F279EAB/connected
unique_id: Aubess_Plug_Tasmota_01_sensor_1
qos: 1
device_class: current
state_topic: obk1F279EAB/current/get
state_class: measurement
platform: mqtt

openshwprojects, 2022-10-23:
You can certainly display the values by hand, just like in this topic: https://www.elektroda.com/rtvforum/topic3887748.html
but the HA **discovery** should work automatically. Thoughts, @iprak  ?

iprak, 2022-10-23:
My understanding from this wiki was that `device_class` should take care of adding the matching unit but apparently it does not.
[image]

I admit I only verified the icon which is more prominent. I will add units to those 3 sensors.

iprak, 2022-10-24:
If you take the latest build and then invoke HomeAssistant Discovery, then you should start seeing units.

daniel-hofer, 2022-11-01:
Thank you for the fix, insatlled now the latest release and can see the units now in homeassistant.
Great work, thanks a lot!

closed 2022-11-01 by daniel-hofer