openshwprojects/OpenBK7231T_App #323: BL0937 socket WiFi(?) crash + strange UI bug

leoshusar, 2022-10-22:
Hi, I have experienced some outage with my BL0937 socket.
I set it up at Friday, everything was going great and then in Saturday I noticed missing data in Home Assistant.
I tried to look up what's happening but it wasn't connected to WiFi at all, it disconnected after almost 16 hours:
[image]

Unfortunately I didn't try to press the socket button to see if it only lost WiFi or if it crashed completely.

But I have now noticed this on the main page:
[image]

It now has 6 hours of uptime, maybe there is some corrupted memory somewhere?
Sometimes with the UI data update it changes to this:
[image]

And now when the device turned off thus stopped pulling current, it's completely normal:
[image]

So:
- when the device is idle, it's normal
- when the device pulls ~10 - <~100 W, it's the first screenshot
- when the device pulls more, it's the second

But when I was calibrating it, I used 50W light bulb and a heat gun with variable power up to 1 kW and this was not happening.
Other parts of the UI seem to be alright, I haven't noticed anything similar.

I was also thinking if #258 might be related. If it happens again today, I'll try to check if it's dead or alive with just WiFi offline.

valeklubomir, 2022-10-23:
It seems to be Wifi issue. Currently working on it.
Also UI bug is worked on.

openshwprojects, 2022-10-23:
@leoshusar we have tested some things with @btsimonh  and determined that thread stack size might be too small in some cases (I suspect it causes addEventHandler and addLog %f crashes), now it's increased, can you recheck if behaviour is still the same?
NOTE: I am not saying that it fixed it, it's just worth rechecking.