openshwprojects/OpenBK7231T_App #324: Stack sizes for HTTP (and some threads?) are now bigger

openshwprojects, 2022-10-23:
Just a note for developers so we all know what was changed - @valeklubomir   @iprak  - with @btsimonh  we determined that HTTP thread stack size was not enough for execution of some script commands, like "addEventHandler", etc. I will also try to reduce the stack size, but for now, I am increasing it from 2048 to 8192
https://github.com/openshwprojects/OpenBK7231T_App/commit/ef019647cf3e36773f47c29d39989a77a1ae71c3
I also increased timer stack size:
https://github.com/openshwprojects/OpenBK7231T/commit/ef03f0f6145449bab8e16c876f9bc373a43bb881
It also caused: https://github.com/openshwprojects/OpenBK7231T_App/issues/314
@btsimonh  you can close 314 now?

iprak, 2022-10-23:
`configTIMER_TASK_STACK_DEPTH` increases from 3000 to 5000 and not 8192, right?

openshwprojects, 2022-10-23:
@iprak yea, what about it?

iprak, 2022-10-23:
Should be it 8192 ? 

Should I make the same change in W600 or maybe later? The value is very low there (`sdk\OpenW600\src\os\rtos\include\FreeRTOS.h`):

```
	#ifndef configTIMER_TASK_STACK_DEPTH
		#define configTIMER_TASK_STACK_DEPTH 200
		//#error If configUSE_TIMERS is set to 1 then configTIMER_TASK_STACK_DEPTH must also be defined.
	#endif /* configTIMER_TASK_STACK_DEPTH */
```

valeklubomir, 2022-10-23:
Also LWIP stack has thread and configurable stack size. I can look at this also.

closed 2022-11-07 by openshwprojects