openshwprojects/OpenBK7231T_App #325: We need to improve/update LWIP for processing large amount of IR data MQTT

openshwprojects, 2022-10-23:
I am getting ERR_MEM when I hold my remote button (drv_ir.cpp) kinda quickly, device is  a single color LED strip, here is the log:

_Info:MAIN:Time 3447, free 97104, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 3448, free 97104, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 3449, free 97104, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 3450, free 97104, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:GEN:dhcp=0 ip=192.168.0.122 gate=192.168.0.1 mask=255.255.255.0 mac=10:d5:61:07:5a:5b 
Info:MAIN:Time 3451, free 97104, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 3452, free 97104, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 3453, free 85232, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 3/38
Info:MAIN:Time 3454, free 97104, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:EVENT:CMD_AddEventHandler: will expect two args.
Info:EVENT:CMD_AddEventHandler: arg2 = 101.
Info:EVENT:CMD_AddEventHandler: arg1 = 1799.
Info:EVENT:CMD_AddEventHandler: added IR_Samsung with cmd led_enableAll 0
Info:EVENT:CMD_AddEventHandler: will expect two args.
Info:EVENT:CMD_AddEventHandler: arg2 = 98.
Info:EVENT:CMD_AddEventHandler: arg1 = 1799.
Info:EVENT:CMD_AddEventHandler: added IR_Samsung with cmd led_enableAll 1
Info:EVENT:CMD_AddEventHandler: will expect two args.
Info:EVENT:CMD_AddEventHandler: arg2 = 96.
Info:EVENT:CMD_AddEventHandler: arg1 = 1799.
Info:EVENT:CMD_AddEventHandler: added IR_Samsung with cmd backlog led_enableAll 1; add_dimmer 10
Info:EVENT:CMD_AddEventHandler: will expect two args.
Info:EVENT:CMD_AddEventHandler: arg2 = 97.
Info:EVENT:CMD_AddEventHandler: arg1 = 1799.
Info:EVENT:CMD_AddEventHandler: added IR_Samsung with cmd backlog led_enableAll 1; add_dimmer -10
Info:MAIN:Time 3455, free 96832, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 3456, free 96832, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 3457, free 96832, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 3458, free 96832, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 3459, free 96832, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 3460, free 96832, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:GEN:dhcp=0 ip=192.168.0.122 gate=192.168.0.1 mask=255.255.255.0 mac=10:d5:61:07:5a:5b 
Info:MAIN:Time 3461, free 96832, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 3462, free 96832, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 3463, free 96832, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 3464, free 96832, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 3465, free 96832, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 3466, free 96832, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:IR:IR decode returned true, protocol 18
Info:IR:Address=0x707
Info:IR: Command=0x60
Info:IR: Raw-Data=0x9F600707
Info:IR: 32 bits
Info:IR: LSB first
Info:IR:IR decode returned true, protocol Samsung (18)
Info:EVENT:EventHandlers_FireEvent2: executing command backlog led_enableAll 1; add_dimmer 10
Info:CMD: enableAll (led_enableAll) received with args 1
Info:MQTT:Publishing val 1 to obk61075A5B/led_enableAll/get retain=0
Info:MQTT:Publishing val 100 to obk61075A5B/led_dimmer/get retain=0
Info:IR:IR fire event took 4ms
Info:IR:Received cmd 00000060
Info:IR:IR decode returned true, protocol 18
Info:IR:Address=0x707
Info:IR: Command=0x60
Info:IR:Repeat
Info:IR: Gap 35900us
Info:IR: Raw-Data=0x9F600707
Info:IR: 32 bits
Info:IR: LSB first
Info:IR:IR decode returned true, protocol Samsung (18)
Info:EVENT:EventHandlers_FireEvent2: executing command backlog led_enableAll 1; add_dimmer 10
Info:CMD: enableAll (led_enableAll) received with args 1
Info:MQTT:Publishing val 1 to obk61075A5B/led_enableAll/get retain=0
Info:MQTT:Publishing val 100 to obk61075A5B/led_dimmer/get retain=0
Info:IR:IR fire event took 4ms
Info:IR:Received cmd 00000060
Info:MAIN:Time 3467, free 96832, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:IR:IR decode returned true, protocol 18
[... 148 lines cut ...]
Info:IR: Raw-Data=0x9E610707
Info:IR: 32 bits
Info:IR: LSB first
Info:IR:IR decode returned true, protocol Samsung (18)
Info:EVENT:EventHandlers_FireEvent2: executing command backlog led_enableAll 1; add_dimmer -10
Info:CMD: enableAll (led_enableAll) received with args 1
Info:MQTT:Publishing val 1 to obk61075A5B/led_enableAll/get retain=0
Info:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:Publishing val 0 to obk61075A5B/led_dimmer/get retain=0
Info:MQTT:Publish err: ERR_MEM aka -1
Info:IR:IR fire event took 3ms
Info:IR:Received cmd 00000061
Info:MAIN:Time 3475, free 96832, MQTT 1(286), bWifi 1, secondsWithNoPing -1, socks 2/38_

here is my script:
_addEventHandler2 IR_Samsung 0x707 0x65 led_enableAll 0
addEventHandler2 IR_Samsung 0x707 0x62 led_enableAll 1
addEventHandler2 IR_Samsung 0x707 0x60 backlog led_enableAll 1; add_dimmer 10
addEventHandler2 IR_Samsung 0x707 0x61 backlog led_enableAll 1; add_dimmer -10_

Fix considerations:
1. LWIP update - @valeklubomir I heard you're onto it, right?
2. Do not publish stuff several times per second if values does not change. Currently, led_enableall publishes every time so with IR we can get SEVERAL led_enableAll with value 1 publishes per second. We can cull it.
QUESTION: do we cull repeated publish "by hand" in led_driver (manually), or do we add some kind of generic mechanism for MQTT with a timestamp and value, that will automatically cull publish values (certain publish values) if they are republished again in the same second? **I will look into it now.**
3. merging QuickTickc (that calls IR) with main loop could help here?

valeklubomir, 2022-10-23:
@openshwprojects LWIP update is in progress

btsimonh, 2022-11-04:
it's pretty damn good now, so closing

closed 2022-11-04 by btsimonh