openshwprojects/OpenBK7231T_App #329: HomeAssistant Discovery / consumption_stats generates error

iprak, 2022-10-24:
The `consumption_stats` generates this error at HomeAssistant end

`
2022-10-24 09:35:07.592 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'device': {'identifiers': ['enbrighten1_8C000000'], 'name': 'enbrighten1', 'sw_version': '1.0.0', 'manufacturer': 'Beken Corporation', 'model': 'BK7231T', 'configuration_url': 'http://192.168.1.211/index'}, 'name': 'enbrighten1 consumption_stats', 'availability_topic': 'enbrighten1/connected', 'unique_id': 'enbrighten1_8C000000_sensor_5', 'qos': 1, 'device_class': 'None', 'unit_of_measurement': 'Wh', 'state_topic': 'enbrighten1/consumption_stats/get', 'state_class': 'measurement', 'platform': 'mqtt'},)
`

The call stack mentions this
`
raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: expected SensorDeviceClass or one of 'apparent_power', 'aqi', 'battery', 'carbon_monoxide', 'carbon_dioxide', 'current', 'date', 'distance', 'duration', 'energy', 'frequency', 'gas', 'humidity', 'illuminance', 'moisture', 'monetary', 'nitrogen_dioxide', 'nitrogen_monoxide', 'nitrous_oxide', 'ozone', 'pm1', 'pm10', 'pm25', 'power_factor', 'power', 'pressure', 'reactive_power', 'signal_strength', 'speed', 'sulphur_dioxide', 'temperature', 'timestamp', 'volatile_organic_compounds', 'voltage', 'volume', 'weight' for dictionary value @ data['device_class']
`

I think this is because `None` is not a [valid device_class](https://www.home-assistant.io/integrations/sensor/#device-class). We probably should just not assign it in that case.

#296

valeklubomir, 2022-10-24:
I expected that None is valid entry for generic sensor. We can leave it empty. 
Question is if dev_cla is optional or mandatory.

iprak, 2022-10-24:
device_class is optional (https://www.home-assistant.io/integrations/sensor.mqtt/). It automatically sets thing like icon in the card. I can see it also being used to present value differently

closed 2022-10-25 by iprak