openshwprojects/OpenBK7231T_App #335: Slow OTA

valeklubomir, 2022-10-25:
After last commit with OTAThreaFix, OTA is very slow, takes 1-2 minutes to complete.
Previous version took less than 20 second to complete.
Creating OTA progress status on front page. Any other ideas?

openshwprojects, 2022-10-26:
@btsimonh  added a rtos delay in OTA because he has realized that in rare cases, a long RTOS function without any sleeps can cause memory issues (or something like that)

I'd suggest to change that delay to a shorter one, but still keep it.

valeklubomir, 2022-10-26:
I know. I have added progress information to main page to see if download was started, instead messing with sleep function. 
Without any notification, it was little confusing, that nothing is hapening.

openshwprojects, 2022-10-26:
I think we could reduce that delay 10 times or so. Just need to check.

iprak, 2022-10-28:
Sometimes, restarting messaging doesn't appear although the device does restart. Could that be due to this change too?

btsimonh, 2022-11-04:
be aware:
if you OTA within the first 30s, it will write to flash from another thread (at boot complete), and this may make the OTA area corrupt.  (note I know nothing about value permenant storage - this could also interrupt OTA).
Usually, this just means it boots back to original firmware....
@iprak - yes, I did sometimes get to OTA effectively over but with no reboot message, and sometimes no reboot.
Reboot is caused by Webapp as a separate REST call ...  I AM seeing some rest calls delayed/timing out from webapp (look at chrome debug), so I think even now we have some HTTP issues.... (even on 1 .123).
However, I do feel that the lastest sprintf teamwork should give us more stability generally :)

The delay in the OTA stuff was because OTA was starving the IDLE thread.  No idle -> no freeing of thread stacks.  8k stacks on HTTP -> quick death if the stacks are not freed.... (notice the new idle spincount in the msgs every second.).

btsimonh, 2022-11-04:
we are back on the edge:
there logs came out all together (i.e. not 1s apart), because the log thread was blocked, because no idle:
```
Info:MAIN:Time 32847, idle 106321/s, free 74232, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 6/38
Info:MAIN:Time 32848, idle 0/s, free 74232, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 6/38
Info:MAIN:Time 32849, idle 485/s, free 74320, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 7/38
```