openshwprojects/OpenBK7231T_App #336: BK7321T Plug sometimes reports irrational high values (around 10kW)

mariusbach, 2022-10-25:
Guys, sorry for opening another one, but: I use a few energy monitoring plugs/sockets which 2-3 times per day report a power draw of 10.8kW or 8.2kW or similar. These occur not in sync with wifi connection losses.

As such a power cannot even be drawn from a household plug, something is going on. I'll drop any such high numbers in my upward data collection system (node-red and influxdb) but wanted to let you know about this.

mariusbach, 2022-10-25:
The error always occurs during minimal load of <5W.

For a [similar tasmota-based plug](https://www.aliexpress.com/item/4001230982267.html) there is the following advice:

```
Notice:
HLW8032  occasionally report invalid power measurements for load values below 5W. During this situation it sometimes reports a valid load. By setting SetOption39 to 128 (default) it must read at least 128 invalid power readings before reporting there is no load.
To discard all loads below 6W simply set SetOption39 1 (0 will reset to default on next restart) so it will report no load below 6W.
```

Although we might not use this chip, it may be a similar problem. I'm not sure though, how their firmware would know that it's an invalid reading. Perhaps it's just above 10/16A and then considered as invalid.

valeklubomir, 2022-10-25:
Working with BL0937 and BL0942 devices, so far this situation did not occured yet. 
But most of these device are used for max output 3680W. So some kind of limiter could be implemented.
So far values <5W work normally. I am testing it with mobile charger and charging modile phone.
Just curious to see how much electricity cost for using modile phone. :-)

mariusbach, 2022-10-25:
Got you a pic. The high spikes above 200W are the ones going to around 10kW. It seldom but it messes up the daily average value. I now filter those out from getting into the database in the first place.
[image]

leoshusar, 2022-10-25:
@valeklubomir OT: the BL0937 doesn't have enough detail for measuring very low values (don't know about 0942), for me it jumps by 1.2 W
[image]

So it might be ok for measuring mobile charger, but once it goes to low power mode (for example charging finished), you won't get those milliwatts flowing through.

valeklubomir, 2022-10-26:
I am fully aware or this. For power minimum step is ~1.5W. Any consumption below this is not measurable. But most charger especially USB C are designed to draw minumu 10W and up to 40W. My data curve looks similary to yours, when it is finished it drops to 0W-1.5W

openshwprojects, 2022-10-26:
I would be suprised if BL0937 or any smart device could measure 0.5W or lower values. I have been testing dedicated "power meters" from china and they all struggle under 1W or 0.5W or so.

konsulten, 2023-04-10:
I have some similar issue with my metering plug were voltage drops to 0 and power goes bananas. 
[image]