openshwprojects/OpenBK7231T_App #340: BL0937 Report period

fjuliofontes, 2022-10-26:
Hi guys, 

This is not an issue but I was trying to understand the logic behind the power consumption IC. I already took a look at the BL0937 demo code but I did not understand the triggers.

From the data that I'm receiving I see a pattern of 6 seconds. Is this right? Will the BL0937 reader report the measurement every 6 seconds or there is another logic behind it? Like seeing if the value is different from the past one, ...?

Thank you,

valeklubomir, 2022-10-26:
BL0937 is producing only pulses from CF/CF1 pins. Pulses are counted with monitoring period 1 second and calculated to respective values for Voltage, Ampere or Watts. Publishing to Cloud occurs with some limits. There is thresholds for changes, so same/similar values are not published repetitively. And at least once per 60 samples (60 seconds) is forced sending.

fjuliofontes, 2022-10-26:
Hi @valeklubomir,
Thank you for the explanation.

So in this example:
 * Machine idle (0A consumption)
 * Machine started (5A consumption)
 * Machine stopped (0A consumption)

Can I assume that the event sent from the idle to start mode and the event sent from the running to stop mode will be within 1 sample (1 second)? Will it have 1 second precision for the "running period" calculation?

valeklubomir, 2022-10-26:
Theoretically yes. But I would count for 1sec per start and stop.
You also may not see immediately rise from 0A -> 5A or fall. 5A -> 0A.
If start is not aligned to measurement you will see rising/falling value.