openshwprojects/OpenBK7231T_App #341: Button long press interval

leoshusar, 2022-10-26:
Hi! Is it somehow possible to set seconds before action trigger on button long press? I looked at the OnHold event handlers code but I didn't see a way to set seconds, so maybe take this as a feature request.

Why - I would like to modify my Tuya dimmer switch which only has short press on / off actions. I noticed that when I hold the switch for X seconds, TuyaMCU resets it's internal timer and ignores this press completely.
This would allow me to solder wire from button to the BK chip and abuse that behavior to trigger my own action on long press :)

Tasmota does this using [SetOption32](https://tasmota.github.io/docs/Commands/#setoptions:~:text=1%20%3D%20Disabled-,SetOption32,-Number%20of%200.1), so something similar would be enough for me.

openshwprojects, 2022-10-26:
So you want to modify BTN_HOLD_REPEAT_TICKS from https://github.com/openshwprojects/OpenBK7231T_App/blob/main/src/new_pins.c ? Edit the time between hold events?
Or do you want to add a delay between button down and FIRST hold event? Edit BTN_LONG_TICKS ?

leoshusar, 2022-10-26:
I think BTN_LONG_TICKS will be it.
When I press and hold button, I want to get single event after specified time, e.g. 3 seconds. This will change my lights from white mode to RGB mode (or the other way) and TuyaMCU will ignore it.

openshwprojects, 2022-10-26:
Ok wait 30 minutes

openshwprojects, 2022-10-26:
Here it is, can you check if it works?
https://github.com/openshwprojects/OpenBK7231T_App/commit/55e61118c243abe2f53fdfec2a9a8184418a8b96
It saves automatically in flash. The defaults are set also if given time is 0 (to avoid issues with older configs)

leoshusar, 2022-10-26:
It works great, thank you!

closed 2022-10-26 by leoshusar