openshwprojects/OpenBK7231T_App #343: OTA can fail if other flash operation happens during OTA

btsimonh, 2022-10-27:
When using webapp OTA, the flash write is not thread protected, so if, for example, we complete boot and write the fact to flash during OTA, the OTA can fail.

It does not cause damage to the device, but it may boot back to original firmware rather than new firmware.

Fixing should consist of mutex protecting flash operations, and making sure they are short and atomic - i.e. getting the main thread stuck for the whole of the OTA would not be a good thing.

openshwprojects, 2022-10-27:
Maybe you don't even need a mutex. Just make a flag that is being set while doing OTA (a global variable) and for example, make CFG_SavePendingChanges check for pending changes every second and save them just really when the OTA flag is not set.

FlashVars could do the same. Something like if(bPendingChanges && BOtaNow == false) { SaveFlashVars(); } would solve.

But I must admit, that's a nice found, so far I experienced a strange OTA fail only ONCE in about 5 months of testing.

valeklubomir, 2022-10-27:
I have added function ota_progress, which return state of OTA or -1 for inactive OTA.

btsimonh, 2022-10-27:
personally, I think we should mutex all our flash operations, and make sure they are setup, done and completed within the mutex.  But each one should only act on one sector...
Note: LFS + OTA is probably a VERY bad idea (same flash area), so LFs should be disabled at the start of OTA.  Since we'll reboot shortly after, should not be an issue to permenantly disable.

valeklubomir, 2022-10-27:
I think flashVars write are secured with GLOBAL_INT_DISABLE/ENABLE we should just copy it to FlashConfig.