openshwprojects/OpenBK7231T_App #353: [Suggestion] Wi-Fi reset to "open" after very long button press, as recovery method

zopieux, 2022-10-29:
Hi there,

Thanks for maintaining OpenBK, I've painfully serial-flashed all 10 of my cheap Tuya switches but it was well worth it. :)

I have a suggestion: it's a bit too easy to enter the wrong Wi-Fi credentials and make the device unusable, necessitating a reflash. This can be super annoying depending on the flash method. Since we already have code for handling long presses, I'd suggest implementing a Wi-Fi reset feature that returns the Wi-Fi to the initial "open AP" state, so people can recover by connecting to the open AP.

To prevent accidental resets, I'd suggest the following state machine:

* button pressed for more than, say, 6 seconds
* LED starts blinking quickly to warn reset is imminent
* LED continues blinking quickly for, say, 4 more seconds
* if the button is still held after 6+4 seconds, LED blinks *very* quickly for a 2 seconds, Wi-Fi is reset, device reboots
* if the button was released within these 4 grace-period seconds, do nothing, and don't register any button press

iprak, 2022-10-29:
I would leave out the LED part, most of the cheaper switches have LED tied to relay.

How about this?
[image]

valeklubomir, 2022-10-29:
SAFE mode requires to restart device 5 times and do not let it run more then 30 second each, then device goes automatically to AP mode and allow change of configuration. Used it few times.

But for example I have TuyaMCU based touch button shield, which would require disassembling when trying invoking SAFE mode, by rebooting it quickly. Which is hard when mounted on the wall. 

So this functionality could be usefull not only for button, for different types also.
But the condition for changing to AP mode or reboot or factory reset should be more strict.
double click or 1sec press is easy and could lead to accidental executions.
Long press should be always more than 5 sec. 

TuyaMCU shield which I have has build in commands from original application.
For example start pairing mode:
click 5 times and 6th press and hold 5sed, then special tuya protocol message is send.

valeklubomir, 2022-10-29:
Some Touch Shields are also capable to pair RF433MHz remote controls. Which also support two more commands:
Pair with remote, clear remote pairing. Again different button combined actions.

leoshusar, 2022-10-29:
Tasmota has 40 seconds long press (10 times the long press length) for recovery, IMO I wouldn't use anything shorter than 30 seconds.

I would even like to have the 5 times restart configurable. We had small power outage yesterday and the electricity went off and on a few times - all my stupid Tuya bulbs went into pairing mode. Great disco wake up in the morning.
It's also quite normal when the electricity company tries to identify problem on wires, for example short. They turn on some part of the grid, then another part, and see which trips off, so the electricity can blink like this.

zopieux, 2022-10-29:
> I would leave out the LED part, most of the cheaper switches have LED tied to relay.

Sure, whatever works, just a nice feedback.

> How about this?

Isn't double click a meaningful bind action already? Binding it to something as disruptive as "AP mode" sounds chevalier.

From the feedback we got in this thread, I'd just go with a very long press then, 40 seconds sounds fine to me. Not sure if it should go full factory reset or just AP mode, both sound fine.

schneemaier, 2023-05-21:
Hello,

What is the status of this feature? 

I ran into the issue today: I ha wifi config changes and during that time 2 of the wall switches froze. They were not reacting to any button press, so I had two option: either open the wall switches and disconnect power or turn of the breaker which supplies the power to these and other light circuits. 

I think we need 3 features if possible:
1. Reboot via the main button
2. Reset Wifi via the main button 
3. Reset to factory defaults via the main button

Thanks

openshwprojects, 2023-05-21:
Hello @schneemaier , I had some troubles with that feature on Tasmota, where wrong configuration (testing configs) led to device going into AP mode, but what about script approach?
You can use addEventHandler with OnHold and command to start AP mode, @schneemaier , would that fit your use case? I can give specific code snippet, if needed, but please see our docs, autoexec examples, etc.

You can also set WiFi ping watchdog to reboot devices once they run into some kind of network issue.

Also please update your devices, newer builds auto-reboot when they detect memory problem

schneemaier, 2023-05-21:
Hi @openshwprojects , thanks for the reply. I have updated both switches yesterday after i rebooted them to the latest firmware (they were on a build from ~February). 

```You can use addEventHandler with OnHold and command to start AP mode, @schneemaier , would that fit your use case? I can give specific code snippet, if needed, but please see our docs, autoexec examples, etc.``` - I would appreciate if you could provide a code snippet. I am pretty sure it is simple, but have very limited time to dig into the documentation. 

I guess the command i need would be ```addEventHandler OnHold <button pin> <command>```, but not sure:
- what is the AP start command 
- if there is a way to specify the onHold time
- for rebooting the device i could use ````addEventHandler OnHold <button pin> reboot```?

Thanks

voodoopt, 2024-03-11:
Yes this option would be a good thing.
Whenever there is a problem with the wifi credentials, being able to reset it in some way or switch to AP mode would be great!