openshwprojects/OpenBK7231T_App #355: How to use the CloudCutter images?

TurkeyMan, 2022-10-29:
I see among the release binaries are some images that are apparently for use with the Tuya CloudCutter exploit, but after a bit of googling, I can't find any confident instructions how to make use of it. Possible to include (or link to) clear instructions to have a go at this method?
I have 2 devices which are welded shut, and I would need to destroy the casing to access the chip to break-out, so I'd be interested to try out the cloudcutter exploit as an alternative to destroying them.

leoshusar, 2022-10-29:
CloudCutter has instructions for flashing [here](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/INSTRUCTIONS.md).
You first have to run the `detach` script and if it's successful, you can use the `flash` script.

ReanuKeeves01, 2022-10-29:
@leoshusar to be fair, i've never ran the `detach` script myself. Always went straight for the `flash` script and pretty much managed this way.

GravityRZ, 2022-11-04:
so you do not need to use detach first and after that run_flash but can straight got flashing a fully operational tuya light.

i noticed that after detaching and running the run_flash it basically does the same(asking to power the unit off/on, waiting for ssid)

good to know when i try a second bulb

please make sure you use the right firmware

GravityRZ, 2022-11-04:
regarding getting tuya cloudcutter to work

i used a fresh ununtu 20.04 image on USB
updated packages
upgraded packages(important because cloudcutter does the same and in my case docker stopped working)
install docker (google ubuntu docker and use the  digital ocean link)
test if docker really works
in my case i had problems with the storage driver because i run from usb flash
fixed it by using this link
https://docs.docker.com/storage/storagedriver/device-mapper-driver/

after that run the detach or the flash script