openshwprojects/OpenBK7231T_App #357: Imprive UART install script

TurkeyMan, 2022-10-29:
Disclaimer: I'm obviously not competent, but better instructions may help myself and others succeed.

I've tried the UART method on my device via windows with 2 different USB->TTL adapters, and also with my pi4. My failure was complete when I accidentally broke the solder pads off from the TX/RX pins from stuffing around for so long.

On windows, I tried using BKwriter and it got past the init waiting step and errored at 2 bars saying it was erasing every time with both USB devices. (some people reported different success with different TTL adapters)
I moved to the pi4 running pios and using the python script, because it seemed a better option for a more mature install process with hopefully some error checking along the way. It never got past the initial waiting step.

Here's some improvement opportunities I can suggest from my failure:

1. Is there some way prior to starting that the script can confirm valid UART communication in both directions?
For example, I spent a long time questioning if TX<->RX + RX<->TX, or TX<->TX + RX<->RX... I didn't feel like I could find a concrete reference for this, so I just tried everything both ways. I also had a wonky solder join on the 7231T's RX pin, which I was never sure if may be causing a problem or not.
If the script could confirm correct di-directional communication on the first step, that would help me to diagnose the problem beyond the UART wiring.

This kind of rookie error could easily be resolved with one line of documentation, and/or an error check at the start of the tool script that UART communication with the chip is working correctly.

2. The script example shows to give the com port by name, it suggests `COM3` as an example, but linux doesn't have devices anything like this. I spent ages trying to work out how linux com ports are identified. While googling around I found some references to things like `/dev/serial0`, but there was nothing like that on my pi. I eventually settled on `/dev/ttyAMA0`, which I still have no idea is correct or not. Somewhere said it may be TX/RX, or it may also be bluetooth...
A few extra lines of text on this detailing different common systems would help a lot.
I never got the python script to continue bast the initial waiting state, and I never knew if this might be the problem?

3. If it's possible to detect the communication is working (#1), why not just scan all UART's for the host rather than requiring the user to specify the port? If the device is boot up with default tuya/smartlife firmware, will the com port respond with anything that can be detected? It would be super cool to just scan all com ports on the system and look for devices that appear to be 7231's.

The script could scan, find the device, inform the user the device it found and if to continue using that port, then prompt the user to reset the device and auto-continue when it detects the bootloader??

iprak, 2022-10-29:
I can confirm a few things.

* It is always TX (device) to RX (adapter) and RX (device) to TX (adapter).
* Do make sure that Gnd (adapter) is connected to the Gnd (device).

Connection confirmation:
* I always do a quick a continuity check with multi-meter. If there is some solder between pins, then it could draw too much power and ruin my power.
* A good way to confirm the connections would be to start a serial monitor. I have used `screen` on Debian. Just power the device and you should see some logging in the serial monitor.

I think some sort of check might be possible if you are using SDK but not production devices.

deepcoder, 2022-10-29:
I have found this simple cli terminal program to be very helpful to have around, bootterm. bt -l when you have too many usb uart plugged in ;-) :

https://github.com/wtarreau/bootterm

ferbulous, 2022-10-31:
> 2\. I never got the python script to continue bast the initial waiting state, and I never knew if this might be the problem?

For me It's about timing as well, you need hit flash right after you powered the uart to the beken chip, if that doesn't work just keep power cycle or ground the CEN pin.
As for ripped tx/rx pads, I'd would recommend using a jig to avoid soldering unless necessary

heino42, 2022-11-05:
Indeed using a jig helped me a lot - great description can be found here (using pogo pins and a 3D printer): https://youtu.be/imKZbhJ8lvU
The chip layout should be mentioned (if you use a jig, of course you will have to mirror the pins). 
After pulling CEN to GND the flashing worked perfectly.

iprak, 2022-11-05:
And here is my monstrosity ... lol
[image]