openshwprojects/OpenBK7231T_App #358: Improve the SPI install script

TurkeyMan, 2022-10-29:
Right now it appears to be just a hacky script you used for your own device one time. It includes commented out lines, hardcoded image names, include a mysterious `REST.bin` image.

Having broken the TX/RX pads off my 7231T and no way I can connect the wires, I figured I'd give SPI a go using my pi4.
I wired up the SPI breakout and connected it to the PI. I activated SPI on the pi (as far as I can tell), but I have no idea if they're talking or not... this seems like something the script could easily attempt to confirm at the head of the script, and notify the user if SPI is not working, or not wired correctly.
It would be ideal to improve the script with an option to supply the image file, and also to distinguish between a normal flash and the 'restore' mode described?
Please clarify the QIO vs UA image dichotomy visible in the script, and that REST thing.

HumbleDeer, 2022-11-04:
Can you specify which script file you are referring to?

TurkeyMan, 2022-11-13:
I mean the script located here: https://github.com/openshwprojects/BK7231_SPI_Flasher

I note that this also exists: https://github.com/OpenBekenIOT/hid_download_py/blob/master/spiprogram
It makes sense to have an SPI flash tool together with the other flashing tools, but the readme says to refer to https://github.com/OpenBekenIOT/hid_download_py/blob/master/SPIFlash.md for SPI instructions, and that document doesn't say anything at all about how to use this `spiprogram` script.

TurkeyMan, 2022-11-13:
I also wonder why those install scripts are hosted as peoples private repos. One is a fork of the other, and not clear which one is the 'current' or 'official' one... why doesn't the org take ownership of that repo and make sure to advertise which revision is 'current'?

HumbleDeer, 2022-11-14:
Hey Manu!

If you're interested in keeping the most up to date version on the repo for
OpenBK listed, you're welcome to contribute to the repo. There's no
requirement except for acceptance of your code to the repo. :)

Happy contributing!

Cheers

btsimonh, 2022-11-14:
@TurkeyMan - the main app repo is in @openshwprojects 's repos.
hid_download_py was something we found, and we created OpenBekenIOT as an organisation to share the code through (it's a fork of a fork, but the OpenBekenIOT should be considered master).
The webapp is also in OpenBekenIOT, and is a fork of mine in btsimonh.  but the OpenBekenIOT version should be considered master.
@openshwprojects also has all the SDK repos.....
https://github.com/openshwprojects/BK7231_SPI_Flasher is a modified version for SPI which should work better (I have not had to use it yet!).  If you will SPI, please document your experience, and submit the documentation.

TurkeyMan, 2022-11-22:
I tried your simple script, but it didn't actually work, gave an error to the effect of "PB not in GPIO set".

[image]