openshwprojects/OpenBK7231T_App #360: Python 2 or 3?

TurkeyMan, 2022-10-29:
I'm not a fan of python, and I'm just trying to follow the python instructions; but every time I see python instructions it's unclear how to follow them. Some instructions say `python3 args`, and other instructions say `python args`.... In other places `pip install xyz` and other places `pip3 install xyz`...
Seems python3 is used here... but is that true for all scripts and all contributors?
Can we please be consistent? Are these scripts all python 3 scripts? Is someone here just aliasing `python` as `python3`? That doesn't seem to be default system configuration, when I `python -v` on pios, I get python 2.
It saw some nonsense about `python-is-python2` in the package manager. Checking my PC, the same thing says `python-is-python3`...
I guess the python ecosystem has ruined this clarity themselves, but it'd be good if this nonsense didn't infect the instructions given surrounding this community. Consider making sure all instructions in all locations are explicit?
The point of these scripts is to be run by lay-folk, not programmers who know what python is or why there are 2 versions, and how to disambiguate, etc. Instructions need to "just work".

iprak, 2022-10-29:
What instructions are you referring to?

TurkeyMan, 2022-10-29:
Nothing specific, just everything everywhere. This is more of a "please keep this in mind, and make corrections when encountering inconsistency" kind of thing.

I'll update this ticket with examples as I encounter them.

iprak, 2022-10-29:
For SPI, I did alias the commands - https://learn.sparkfun.com/tutorials/python-programming-tutorial-getting-started-with-the-raspberry-pi/configure-your-pi

TurkeyMan, 2022-10-29:
> For SPI, I did alias the commands - https://learn.sparkfun.com/tutorials/python-programming-tutorial-getting-started-with-the-raspberry-pi/configure-your-pi

Hmm... arbitrarily aliasing the commands sounds bad... surely that's just likely to mess up anything else that might be running on your system?
Why not just be explicit in instructions here?

iprak, 2022-10-29:
Yes it could but I was using a throw away setup. 

You are very welcome to adjust the documentation. My experimentation with SPI was a couple weeks ago and I could not get the working hardware end working but will give it a try again.

TurkeyMan, 2022-10-29:
Yeah, I wired it up and had a quick crack, but it didn't "just work", and I'm not sure how to diagnose where the problem is. I'd like to think it's possible for the script to integrate basic fault tests.

btsimonh, 2022-11-01:
@openshwprojects did some SPI a few days ago, and may have a cleaner implementation.  Mine was 'hack until it worked'.