openshwprojects/OpenBK7231T_App #365: LSC 2578539-970724 - Correct module config

ReanuKeeves01, 2022-10-29:
Anyone who got this bulb flashed and working mind sharing the right module config? 

[image]

I'm seeing this on the forums, and the auto-config in OpenBK seems to have followed this. It's just only filling the first field, not the actual values. 

Many thanks

ReanuKeeves01, 2022-10-29:
Answering my own question for future reference.

Pin 24 = BP5758D_CLK = Value 0
P26 = BP%&58D_DAT = Value 0

To now make this work in the interface, turn on the following setting:

[image]

As per [forum discussion](https://www.elektroda.com/rtvforum/viewtopic.php?p=20258613#20258613) this would be a nice QoL feature for the less educated user to be automatically switched on when in this configuration.

Bacto, 2022-10-29:
Good find @ReanuKeeves01!
I'm facing a new issue on my side. I can control the bulb using MQTT. But can't add it using the "Home Assistant Discovery" system.

I get "No relay, PWM or power driver running."

[image]

ReanuKeeves01, 2022-10-30:
@Bacto I have another device that also doesn't work with Auto Discovery, same error. 

What worked for me, was adding the device manually in Home Assistant, in the configs by copying the config on yoru screen, into your `configuration.yaml` file.

In case you already have other mqtt entries in that file, your integration may vary slightly.

ReanuKeeves01, 2022-10-30:
Also make sure this setting is on, from Configure General:

[image]

openshwprojects, 2022-10-30:
I think I will just commit a change that will ALWAYS show RGBCW controller for all I2C LED drivers. Then this flag won't be needed...

Bacto, 2022-10-30:
@ReanuKeeves01 thanks for the reply. Unfortunately flag 7 doesn't resolve the problem.

@openshwprojects do you have a clue?

ReanuKeeves01, 2022-10-30:
@Bacto have you tried manually adding the entity/device through your configuration.yaml? And if you did, what were the results or outcome ?

Bacto, 2022-10-31:
Manually adding the JSON configuration to `configuration.yaml` works.
That's why I think it's a OpenBK7231T_App bug. It shouldn't refuse to send the JSON configuration through MQTT.

ReanuKeeves01, 2022-10-31:
@Bacto i only managed to get the auto-discovery properly working for two CW/WW bulbs. The RGB ones I had to manually add via yaml to get everything working.

HumbleDeer, 2022-11-04:
Can you check if a command / terminal logbook with all types of output enabled (eg warnings, errors, info, etc) shows anything related? if so, can you post it?

GravityRZ, 2022-11-04:
i added mine to domoticz but some controls do not work
color for starters do not work
dim works
off work
on does not work

i think because the led_basecolor_rgbcw command is not supported yet.

do you actually have this RGBWW bulb fully working in Homeassistant.
if that is the case it should work in domoticz to

Bacto, 2022-11-04:
Hi,

I didn't got time to look at it but I saw that:

- When flag 4 is enabled:
  - I have light controls in OpenBK7231T_App WEB UI
  - MQTT led_basecolor_rgbcw doesn't work
- When flag 4 is disabled:
  - I don't have light controls in OpenBK7231T_App WEB UI
  - MQTT led_basecolor_rgbcw works

Best,
Adrien

GravityRZ, 2022-11-04:
i will try but seems strange.
looks like they do not want you to change the color from the gui AND homeassistant/domoticz

ReanuKeeves01, 2022-11-04:
@GravityRZ very strange, i can at least confirm that i do have them fully working in both the OpenBK WEB UI, as well as in Home Assistant, with full RGB and CW/WW support. 

What i noticed, is that when i add the new light in Home Assistant, as it is, the controls don't show up properly. I had to 'touch' all the controls once in OpenBK WEB UI once. I.e. Turn the light on/off, change the brightness, change the color temp and set one or two colors... 

After that, the settings suddenly appeared correctly in HomeAssistant instantly. Flag 7 (from the general options, the retain flag) seem to be a must on my side. When i do not use this, after reboots of either OpenBK or HomeAssistant, it looses the controls and i'd have to repeat the steps above to get them showing up again in HA. Not sure if any of this helps, but this was my process to get them fully operating in HA.

GravityRZ, 2022-11-04:
Good info. Will try tonight

GravityRZ, 2022-11-05:
i tired it but is is recognised as RGBWZ instead of RGBWW
do not know what RGBWZ is