openshwprojects/OpenBK7231T_App #366: Can't get RGB lights support with Home Assistant discovery

Bacto, 2022-10-29:
Hi,

I'm new to OpenBK7231T_App and Home Assistant so please pardon me if I didn't understand something right :)

I'm trying to add a RGB light to Home Assistant using the "discovery" feature.

I have the RGB support on the OpenBK7231T_App WEB UI:
[image]

When I click on "Start Home Assistant Discovery" this JSON is sent to MQTT:

```json
{
  "dev": {
    "ids": [
      "led_mood_light_1"
    ],
    "name": "led_mood_light_1",
    "sw": "1.14.86",
    "mf": "Beken Corporation",
    "mdl": "BK7231N",
    "cu": "http://192.168.1.139/index"
  },
  "name": "led_mood_light_1",
  "~": "led_mood_light_1",
  "avty_t": "~/connected",
  "pl_on": "1",
  "pl_off": "0",
  "uniq_id": "led_mood_light_1_light",
  "qos": 1,
  "rgb_cmd_tpl": "{{'#%02x%02x%02x0000'|format(red,green,blue)}}",
  "rgb_val_tpl": "{{value[1:3]|int(base=16)}},{{value[3:5]|int(base=16)}},{{value[5:7]|int(base=16)}}",
  "rgb_stat_t": "~/led_basecolor_rgb/get",
  "rgb_cmd_t": "cmnd/led_mood_light_1/led_basecolor_rgb",
  "clr_temp_cmd_t": "cmnd/led_mood_light_1/led_temperature",
  "clr_temp_stat_t": "~/led_temperature/get",
  "stat_t": "~/led_enableAll/get",
  "cmd_t": "cmnd/led_mood_light_1/led_enableAll",
  "bri_stat_t": "~/led_dimmer/get",
  "bri_cmd_t": "cmnd/led_mood_light_1/led_dimmer",
  "bri_scl": 100
}
```

Seems that RGB configuration is defined to it.

But once on Home Assistant I have only 1 entity which is a power switch and no RGB nor dimmer configuration available:
[image]

I tried a lot of things but can't find why I only have one entity :-/

Any help would be greatly appreciated :)

Thanks,
Adrien

ReanuKeeves01, 2022-10-30:
Well, you should have only one entitiy, seeing it's a single lamp, correct?

When click on the entity, how is the info tab looking? Any controls there? 

One thing i noticed, is that Home Assistant doesn't seem to 'know' what options the bulb has, until i manually adjust them once (on/off, brightness, color, cw/ww through OpenBK itself) - from then HA's INFO tab on that entity would actually look the way it should be. With the right controls. But after restart it would be gone again and i had to redo it.

This setting solved that:
[image]

iprak, 2022-10-30:
I don't remember but HomeAssistant might not RGB controls if the lamp was not turned on. What happens if you clicked on the bulb icon?

iprak, 2022-10-30:
Also does the light entity appear unavailable?

Bacto, 2022-10-30:
Damn it! I had to click on the "brightness" setting to see the RGB setting... So much hours lost!
Thanks to both of you :)

closed 2022-10-30 by Bacto

ReanuKeeves01, 2022-10-30:
Good, you solved it!

Isn't it great spending hours fiddling just to realize it was THAT simple. I had a similar experience last night with controls not showing up correctly in HA. Turned out all i had to do was touch the different controls once through OpenBK. Like switchign it on and off. Increasing the brightness and changing the color. After changing the values through OpenBK once, now suddenly Home Assistant saw the capabilities right and everything worked.