openshwprojects/OpenBK7231T_App #368: drv_bp1658cj debuging help not working

botioni, 2022-10-30:
bp1658cj not working can we have some debuging output from driver and some respons output and cmd 
g_pin_clk and g_pin_data set from cmd command 
somthing like  bp1658cj_g_pin_clk=8 ,bp1658cj_g_pin_data=24.

openshwprojects, 2022-10-31:
We do have logging, I think, but you need to type "logtype 7" or smth in order to display it:
[image]

botioni, 2022-10-31:
I did but all loging is for user input not for respons  from network .
And the problem is thet is not working and I dont know why .
Can be dosent changing gpio pins or somthing else .
Wold be nice to have loging about the pins thet it uses or to set them manually and to semd direct commands to the pins

botioni, 2022-11-02:
Can sombady report thet the driver is working ?
Did sombady test it on other devices ?

botioni, 2022-11-03:
no clock on analyzer (i ended up buying an Cheap logic analyzer SPI i2c UART) can sombady look into the code or at last test if it is working any driver i2c comunications

botioni, 2022-11-03:
driver working properly

botioni, 2022-11-03:
some other bug

closed 2022-11-03 by botioni