openshwprojects/OpenBK7231T_App #369: [Feature request] HTTP basic auth on WEB UI

Bacto, 2022-10-30:
Hi,

Again, thank you for this project!

One bad point is that I'm really unconfortable having the WEB UI exposed without an authentication system.
Anybody on the same network can take control of the device and even worst, update the firmware.

I think adding HTTP basic auth will resolve this and should be "easy" to implement.

Unfortunately I'm not confortable at all with this project (I just flashed 2 devices for now and I'm a total beginner with BK7231x chips) so making a PR is not an option for now :(

Somebody would be able to add this feature?

Best,
Adrien

ReanuKeeves01, 2022-10-30:
This would make a lot of sense and make for a good feature addition. Another valuable addition I thought of would be to have the option to disable the HTTP API. In my case, I'd happily switch this off for most devices since it's all running over MQTT..

valeklubomir, 2022-10-30:
I will put it to my TODO list. NOT exclusively. Any help is welcome. Still enough TODO with stability and MQTT transfer rates.
Cyber security is very important feature.
- Disable WEB UI by startup command or limit WEB UI to selected IP. 
- HTTP BASIC AUTH is not safe enough, password is still sent as plain text, may be plug-in to do HASH in script would help.
- May be SSL for MQTT. I have some experience with ESP32 devices in this matter.

openshwprojects, 2022-10-30:
Yea, this is a good idea, but not a super top priority. One thing to note would be that we also have a raw TCP command server running (if enabled), and that also would require some security.

HumbleDeer, 2022-11-04:
When reading these ideas, things that came to mind:

- setting a flag during boot that needs to be cleared after successful booting first before the HTTP API / Server is stopped or minimized. 
- retaining basic REST API functionality for retrieving status of certain things - handy for a backup solution when HAss or other software loses track due to MQTT issues that can arise
- Password security can be done using simple classic Pub/priv key auth with comparison to a pub key locally (client). Security concerns about cracking this should be handled at the network level first -- unauthorized users shouldn't have access to your network to begin with.

iprak, 2022-11-07:
@openshwprojects @valeklubomir I locally have implemented Digest authentication and am testing it more. I will create a push request at some point to optionally enable it for additional testing.

iprak, 2023-01-13:
I did some work but it only provided some limited coverage for the embedded webApp.

The WiFi password is not the only unsecure aspect. MQTT credentials are exposed. APIs can be invoked without any intervention.

The firmware in mostly suitable for controlled networks. One path I have taken is to put them on a secondary network, so that they cannot be accessed freely. You will however need to route the various ports through firewall.

I can try to reinstate my attempt at some point.

Till then, you can move the devices to a guest network with mqtt port

ph0tic, 2023-04-08:
+1 for this. would love to have it.

castorw, 2023-12-13:
PR in progress: #996