openshwprojects/OpenBK7231T_App #372: [CB2S] Adjust the number of decimal send by energy meter

BugsBunny1403, 2022-10-30:
Hello, i ask if it possible to select number of decimal send by energy meter to mqtt principaly and http web page.
As Tasmota by CurrentRes , PowerRes or VoltageRes.
Sorry for my bad english.

HumbleDeer, 2022-11-04:
Do you mean you'd like to have an adjustable rounding depth?

If so, that should be pretty easy to do by copying over from Tasmota.

btsimonh, 2022-11-04:
the latest version .123 uses a new vsnprintf.
the only issue we have found so far is that BY DEFAULT, %f is 6 decimal places, so %f of float 1.0 is printed as 1.000000!!!!
@openshwprojects - The power readings should probably be changed to %.3f (?) or even %.1f depending upon what they represent?

iprak, 2022-11-11:
My experience has only with 2 devices: TP-Link Kasa power monitoring plugs and [Emoncms](https://emoncms.org/) and they both only report data to 1 place of decimal. I think the accuracy of consumer level monitoring devices is quite low.

I would like to mention that this can be handled at HomeAssistant level as part of `value_template`.

BugsBunny1403, 2022-11-27:
Hello, I would up this request for know if it possible.

torge70, 2022-12-25:
+1 from me.

catalinbordan, 2023-02-13:
+1 from me.