openshwprojects/OpenBK7231T_App #374: Sonoff B05-BL

Thom4fun, 2022-10-30:
Hello, I need some configuration help on a Sonoff B05-BL bulb. It is a new bulb with DIY Mode. The used MCU is an BL602, the LEDs are driven by a BP5758D Chip. (I double checked that by opening the whole module)
Finaly, I was able to flash the chip and configured it by p01 and 02 with BP5758D_Dat and CLK.
I also tried a lot other configurations, including option 4 and 9. But no luck, no light at all.

After I realized that the BP5758D_Map goes quited as command not found, I think that the BP5758 Chip is not includet in BL602 Firmware.
Can you give any help in that case / combination?

Thank you for your already done work, what a great project.

Kind regards
Thomas

openshwprojects, 2022-10-31:
Thank you for reporting. I have not been aware about a BL602 BP5758 device. This is a news to me.

I think I can get a BP driver running easily on BL602 for you. I just need to find my dev board. Please wait several days.

In a meantime, would you be kind enough to make a short information dump and/or a teardown for our global IoT devices list?
The list is here: https://openbekeniot.github.io/webapp/devicesList.html all you need to do is to post the packaging photos (if possible), device photos, model information, shop information, inside photos etc. on our  forums: https://www.elektroda.com/rtvforum/forum507.html

The BL602 is rarely found in IoT devices, so your discovery really matters.

Thom4fun, 2022-10-31:
Thank you for your fast reply

I am new to the thematic of MQTT and all around. So I bought some devices for trying and was shocked what to found on the market.
Fast I found Tasmota to see that the most new devices seams to be not supported anymore. (Or unlucky me, who find the wrong devices.) But anyway, I found yyour great project and I would like to contribute your work.

I will publish what I did and found…

Thank you for your kind help.

Thomas

openshwprojects, 2022-10-31:
We will support all supportable chips. There is no need to worry. Everything will be cloud free and open source.

The "doing teardowns" initiative is required to make knowledge widely available and accessible for everybody.

BL602 is not popular, as far as I know, so it's interesting to see it with BP driver. As said, I will try to enable that one for BL.

How much BL602 devices do yo have?

Thom4fun, 2022-10-31:
I have all together 5 devices. I thought the API will do the job. But the API is so uncomfortable, like each device has his own ID which must be used, that it is nearly useless. So I flashed just one. There is another B05-B (Without “L”), but there is another chip in there. This light give some functions, but do not run well with your firmware.  I will do a tear down as well for this light and will give a better description of the malfunction.

Thanks again for your help.

Thomas

openshwprojects, 2022-11-01:
Thank you for posting teardown, I will add it to our online database.
https://openbekeniot.github.io/webapp/devicesList.html
SV6166F is also new to me, it would be interesting to see it. I will check if this chip has SDK, maybe it's supportable.

Regarding BP5758D - the driver is only doing digital operations on pins, so it should  be easy to just enable it in SDK, i will check.

openshwprojects, 2022-11-04:
@Thom4fun I have enabled BP in BL602

Thom4fun, 2022-11-04:
Cool, thank you. I will have a try right now and will report.

Thom4fun, 2022-11-04:
I am sorry to report, that the test was negativ. No light at all, as well the command BP5758D_Map went returned with command unknown. The index page informed that one driver is active: BP5758D. The web app do not show any slider for white light or RGB. But I think about the pin configuration. The PINS at the WIFI module are labeld PWM1 to 5 where 1 and 2 are connected to the BP5758, what should be GPIO0 and GPIO1. It is why I used P0 and P1. Do I am right? Any other hint what I am doing wrong?

Regards
Thomas

openshwprojects, 2022-11-06:
@Thom4fun as with the remaining drivers, you must check flag - "Show RGBCW controller". Enable this flag and you will get GUI.

Thom4fun, 2022-11-10:
I try to stepp foreward with the driver. The board is labled with PWM1 ... PWM5. I compare that with the Pinout of the BL602. There are five posibilities of available Pins. But what does the label in your software say? Is it GPIO? or is it the true Pin nr?

Thom4fun, 2022-11-10:
Hi Guys! 

I have the next step, bulb is lightening by using the GUI. But I do not have any time left for further testing. I will continue may within one or two weeks.

The command BP5758D_Map is still missing, but it works as well without, just, the colors blue and green are mistaken.
The goal was to put BP5758D_CLK on PIN 2 and BP5758D_DAT on PIN 21. whereby I put PIN2 as channel 1 and PIN 21 on channel 2. Not sure if it is necessary.
I will correct and enrich the teardown and have some tests like MQTT and so on, as soon as possible.

Nice Firmaware, graet job, perfect support. 

Thank you guys....

Regards
Thomas

openshwprojects, 2023-02-22:
Topic is covered on our video:
https://www.youtube.com/watch?v=L6d42IMGhHw&ab_channel=Elektrodacom

closed 2023-02-22 by openshwprojects