openshwprojects/OpenBK7231T_App #377: Is energycounter_clear_date in correct format?

iprak, 2022-10-31:
Seems the sensor `energycounter_clear_date` is emitting date in local toString format.

`Info:MQTT:Publishing val 2022-10-30 12:09:04 to enbrighten1/energycounter_clear_date/get retain=0`

But https://www.home-assistant.io/integrations/sensor/#device-class indicates that `date` with `ISO 8601` format.

@valeklubomir 566bebca1e01515e46049e641a84853f0a6d1429

valeklubomir, 2022-10-31:
Actually I am not sure which format to use, only thing writte there was to use python compatible datetime format. 
whatever it means, because there are multiple format, depending on setting of the other system. 
I was just bold to presume that higher systems with more computational power would autodetect the format and import it correctly. 

If you have an example how it should look like, we can adjust the sprintf formating. 
I think there is no neccessity to implement mutli format libary.

valeklubomir, 2022-10-31:
Would be this format usefull?
2019-09-07T15:50-04:00

iprak, 2022-10-31:
Yes that is ISO 8061.

But I have another question, is the value represented by `energycounter_clear_date` a timestamp or date? if timestamp, then device_class should be timestamp.

Sample timestamp value from Hass:
[image]

valeklubomir, 2022-10-31:
Ok I change it to timestamp.

valeklubomir, 2022-10-31:
Fixed created pull request.

closed 2022-11-04 by iprak