openshwprojects/OpenBK7231T_App #379: [MQTT] We must handle correctly resending MQTT values that changes, let's say, 100 times per second

openshwprojects, 2022-10-31:
I am working on scripting stuff and right now even I got a scriptable dimmer working:

```

// brightness loop
// First makes it gradually brighter, then darker
// Channel 10 is dimmer delta
// Channel 11 is delay step in ms

setChannel 10 2
setChannel 11 25
led_enableAll 1

again:

if $led_dimmer>100 then "setChannel 10 -2"
if $led_dimmer<0 then "setChannel 10 2"

delay_ms $CH11
led_dimmer $led_dimmer+$CH10
goto again
```

the problem is that, with current state of things, that would generate.... well, a MQTT publish every 25 ms, so 40 MQTT requests per second

This is unacceptable.

I think we should alter MQTT publishes system, don't know yet if globally (and do it automatically) or only per LED driver, but the basic driver should be such:
1. Code requests publish of led_dimmer
2. System checks if led_dimmer was published during last 1 second, if not, just publish it
3. If it was published during last second, then just mark as dirty and wait for a second (even if it changes 10 times during that second, ignore it)
4. After second passes, publish the LAST, most up to date dimmer value

I might modify my "deduper" to handle that. This is the only sensible approach I can think off. 

I'd imagine something like:

struct mqttPublish {
     char *lastValue;
     bool bLastValueSend; // if false, then it will get resend after timeSinceLastValueSend is bigger than 1 second
    float timeSinceLastValueSend;
};

Any thoughts? Or maybe is there a better way to handle it? @iprak @valeklubomir  @btsimonh

btsimonh, 2022-10-31:
Personally, I would (for MOST mqtt) set a flag or a time to publish, and then ONLY publish from a single thread (fast tick?).  Then, if a topic is requested twice within a few ms, it would just replace the send timer with a fresh one.
e.g. mark MQTT of dimmer value to send in 100ms.
If a second request 'mark MQTT of dimmer value to send in 100ms' just reset time to send to 100ms from now.  Build data to send at send time (so you get the latest value).
Note: DGR should be immediate.... although you still may want to send from a single thread.

btsimonh, 2022-10-31:
@valeklubomir - I think your mod to mqtt SDK to remove mqtt_output() from end of publish delays send....  i.e. I don;t think we ACTUALLY send unless a puback or a heartbeat occurs?  (observed on N with IR blasting MQTT).  Was this the intent?  or do we need something like "if (mqtt is not sending) mqtt_output()" at the end of the publish fn?

btsimonh, 2022-10-31:
p.p.s. All be aware the the lwip MQTT implementation is actually very efficient in it's buffering.  It uses a ring buffer, and stores the minimum amount of data per queued message.....

openshwprojects, 2022-10-31:
I have a working change for deduper to do exactly what I proposed. Currently with 1 second forced delay if we want to update too fast. I will test it and commit, but it's just a #ifdef , so it will be easy to disable if something breaks.

Of course, we can also migrate to other approach any time, no breaking changes were made by me so far.

valeklubomir, 2022-10-31:
@btsimonh Removing mqtt_output was due to collisions between thread which calls mqtt_publish and thread calling mqtt_poll_cb.  Collision have caused corruptions on socket in TCPIP. IF we want to reach 100 MSG/s we need just to increase buffer to accomdate 100+ msg in ringbuffer and req_list. 
And threads should always call TaskDelay to leave control to other theads handling other operations.

btsimonh, 2022-10-31:
@valeklubomir 
agree on the taskDelay (or rtos_delay_milliseconds) - you saw my one on if idle gets no time, things go horribly wrong?!.
It would be good if all out MQTT was handled in a single one of our threads.

btsimonh, 2022-10-31:
(basically - idle is responsible for freeing thread storage for threads which have finished, so if it does not run after HTTP request completes, 8K of malloc hangs around.  After the fourth http request with no idle, we're into no-memory territory.
I think the 8K stack is not necessary for HTTP thread - but have not had time to throw stackCheck in loads of places to see what we actually need - be aware of this new routine in memory/memtest.c for BK T/N.

valeklubomir, 2022-10-31:
@btsimonh 
TCPIP Network communication thread handles more than MQTT. So why create additonal thread for handling data when already one exist. At end data must still go through single physical device. So two different threads shouldn't try to access HW.
TCPIP thread is designed to handle incomming and outgoing communications.

openshwprojects, 2022-10-31:
"IF we want to reach 100 MSG/s  need just to increase buffer to accomdate 100+ msg in ringbuffer and req_list."

There is no need for that. My script can indeed change a brightness of led_dimmer about 50 times per second, but what is the point of sending that? Just send one value every second and then make sure to send the LAST value when changing stops, so HA knows what is the current (final) brightness of the strip, and that's all.

I am testing my new deduper, it seems to work, but it seems I will have to add a mutex in it. I will commit it soon so you guys can look into it.

valeklubomir, 2022-10-31:
T version tested with LWIP 2.1.3. Created pull request. Up to 20 msg/s is possible.

openshwprojects, 2022-11-04:
@valeklubomir thanks to @valeklubomir work, this should be fixed now. I am closing this issue.

closed 2022-11-04 by openshwprojects