openshwprojects/OpenBK7231T_App #380: All Devices / Configuration settings are not consistently saved

iprak, 2022-10-31:
Devices: All

It seems that one should be calling `CFG_Save_SetupTimer` or `CFG_Save_IfThereArePendingChanges` after setting configuration value but many functions are not doing that. I noticed this in `http_fn_cfg_name`.

openshwprojects, 2022-11-01:
This is obviously a simple mistake. Just add a missing call. Great catch!

PS: We could also just call "CFG_Save_IfThereArePendingChanges" in a main, every second, loop, hm...

but we also might need to check if OTA is running, so we don't write to flash from two threads! @btsimonh

iprak, 2022-11-01:
This was only missing at 3 spots so calling `CFG_Save_IfThereArePendingChanges` would be sufficient.

closed 2022-11-01 by openshwprojects