openshwprojects/OpenBK7231T_App #383: TODO: implement and test SSDP on other platforms

btsimonh, 2022-11-01:
Currently, for platforms other than BK, C files need to be added to the makefile.

We need some help from someone actively using the other platforms to confirm that if drivers/DRV_SSDP.c is added, and 'startdriver SSDP' is run, that the device appears in Windows networking in 'Other Devices'.

once confirmed, the makefile changes can be committed....

ataa, 2023-02-25:
I tried to activate it on BL602:
```

Info:MAIN:Driver SSDP is not known in this build.
Info:MAIN:Available drivers:
Info:MAIN:TuyaMCU
Info:MAIN:, tmSensor
Info:MAIN:, NTP
Info:MAIN:, TESTPOWER
Info:MAIN:, TESTLED
Info:MAIN:, HTTPButtons
Info:MAIN:, BL0942
Info:MAIN:, BL0937
Info:MAIN:, CSE7766
Info:MAIN:, DDP
Info:MAIN:, SM2135
Info:MAIN:, BP5758D
Info:MAIN:, BP1658CJ
Info:MAIN:, SM2235
Info:MAIN:, Bridge 
```

openshwprojects, 2023-02-25:
@ataa for SSDP , one must do that:
https://github.com/openshwprojects/OpenBK7231T/commit/b0f232db5f17b3ac558580767a677f2a2ee6d488
I can try to add it to BL602, waaait...

openshwprojects, 2023-02-25:
ok it works on bl602

ataa, 2023-02-27:
Thanks, it is working.