openshwprojects/OpenBK7231T_App #386: Tunable white configuration.

valeklubomir, 2022-11-01:
Device #1: LSC Smart Bulb 1400 lm/E27/Tunable White
Device is 14W tunable white, with BK7231T and simple current drivers controlled by PWM. 
PWM1 - Cold White
PWM2 - Warm White

Device #2: LED SMART WIFI Ceiling light 
Device is 28 W, 1960 lm, tunable white 3000 – 6000 K controlled by special driver BP5926.
PWM1 - Light Intensity
PWM2 - Light Temperature

"Stupid" question. How to correctly setup this devices?

openshwprojects, 2022-11-01:
Device 1 - just Channel 0 to PWM and Channel 1 to PWM and it should happen automatically? 
Device 2 - as above, but then set flag:
[image]

valeklubomir, 2022-11-01:
[image]
PWM index ?  Is 0 used?  Regular users are not programmers most do not know that 0 can be used.

openshwprojects, 2022-11-01:
The PWM index selection is smart and just checks which index is defined first. If that's 0, then 0 is used. If that's 1, then 1 is used.

valeklubomir, 2022-11-01:
How do I setup that light correctly? After REBOOT/RESTART, I see this:
[image]
But light is off in the real, only after changing any setting brightness or temperature light goes ON.

openshwprojects, 2022-11-01:
@valeklubomir  I saw that, I think it's a little bug, the light is off at the start and somehow the led state seems to be 1, I will look into it tomorrow.
But basically, it should work okay if you just click "Toggle Light" twice. 
THE FIX is most likely just:
[image]

changing it to 0 here. I gotta go now, I will look into it tomorrow morning (in about 14 hours at most from now) and also commit a lightweight vsnprintf

closed 2022-11-03 by valeklubomir