openshwprojects/OpenBK7231T_App #387: Upgrade from .105 to .108 seems to break something - BK7231T

mariusbach, 2022-11-01:
I have upgraded 4 wifi sockets from .105 to .108.

3 have BL942, Periodic Energy Statistics and NTP drivers enabled - all of them are not reachable after the upgrade. No ping possible, not even a reaction to a button press on the device itself. 

However, I can unplug/plug a few times until they start in AP mode; I can connect and use the menus, but I cannot downgrade the firmware because the web app doesn't open. Also I cannot toggle the relais of the plug, not via software and not via hardware device button.

1 socket has no driver enabled - upgrade without issues.

valeklubomir, 2022-11-01:
BK7231T. My device is Light Bulb. I am running NTP without problems.

openshwprojects, 2022-11-01:
First I would like to note that in Safe Mode, the pins are disabled, so it's normal that you can't toggle relays. The idea is that in Safe Mode, you can revert breaking changes and then put back device in normal mode.

mariusbach, 2022-11-01:
> in Safe Mode, the pins are disabled, so it's normal that you can't toggle relays

Yes, that's what I thought as well. Fair enough. 

I'm not sure how to proceed and downgrade via OTA, when I only can connect in AP mode. I don't have an http-server in that AP-mode-network.

mariusbach, 2022-11-01:
> Maybe it's LFS related? Was your autoexec.bat lost?

I don't use the LFS or autoexec.bat. Also I usually press the "quick OTA" button because I don't need to preserve the LFS.

btsimonh, 2022-11-01:
remove mqtt and ntp config in safe mode.  disable other drivers if present?  ensure it's run for >30s, then reboot and see if it comes back on normal wifi?

openshwprojects, 2022-11-01:
@mariusbach are you on T platform or N? @btsimonh  is it possible that N platform packs config struct differently than T? Just a wild guess.

mariusbach, 2022-11-01:
@openshwprojects I'm using the T platform.

> remove mqtt and ntp config in safe mode. disable other drivers if present? ensure it's run for >30s, then reboot and see if it comes back on normal wifi?

This worked! It's back online. I'll try to re-enable the drivers one by one.

mariusbach, 2022-11-01:
I did setup the MQTT server as a first step, no driver enabled. Return:

```
Wifi RSSI: Excellent (-49dBm)
MQTT State: disconnected RES: -6(ERR_VAL)
MQTT ErrMsg: mqtt_client_connect connect failed
```

mariusbach, 2022-11-01:
Once I enable `backlog startDriver BL0942;`, I cannot connect anymore.

btsimonh, 2022-11-01:
we'll need someone who has BL0942 + serial logs.  Or to comb source code changes....
for serial logs, use
logdelay 100
to ensure you see the last log before death.

mariusbach, 2022-11-01:
@btsimonh 
> also, try
> startdriver BL0942
> without backlog, and not in startup command line.

I did execute this as custom command: Device immediately is offline.

mariusbach, 2022-11-01:
I did use the webapp for logging and sent command startdriver BL0942 from the webapp, result:

```
Debug:CMD:cmd [startdriver BL0942]
Info:EnergyMeter:Read ENERGYMETER status values. sizeof(ENERGY_METERING_DATA)=32
Debug:CMD:Adding command EnergyCntReset
Debug:CMD:Adding command SetupEnergyStats
Debug:CMD:Adding command ConsumptionThresold
Debug:CMD:Adding command PowerSet
Debug:CMD:Adding command VoltageSet
Debug:CMD:Adding command CurrentSet
Debug:CMD:Adding command PREF
Debug:CMD:Adding command VREF
Debug:CMD:Adding command IREF
Info:NTP:Started BL0942.
Info:GEN:9 Button_OnLongPressHold
Info:GEN:9 Button_OnLongPressHold
Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0
Info:MAIN:Time 26, idle 700254/s, free 96544, MQTT 0(2), bWifi 1, secondsWithNoPing -1, socks 2/38

```

openshwprojects, 2022-11-01:
@mariusbach there is a "TESTPOWER" driver which acts like a BL0937/BL0942 device.

Can you try again but with "startDriver TESTPOWER" instead of BL0942 and see what happens?

mariusbach, 2022-11-01:
@openshwprojects Ok, done, result:

```
Debug:CMD:cmd [startDriver TESTPOWER]
Info:EnergyMeter:Read ENERGYMETER status values. sizeof(ENERGY_METERING_DATA)=32
Debug:CMD:Adding command EnergyCntReset
Debug:CMD:Adding command SetupEnergyStats
Debug:CMD:Adding command ConsumptionThresold
Info:NTP:Started TESTPOWER.
Info:GEN:9 Button_OnLongPressHold
Info:GEN:9 Button_OnLongPressHold
Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0
Info:MAIN:Time 19680, idle 718739/s, free 96952, MQTT 0(1230), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:GEN:dhcp=0 ip=192.168.110.153 gate=192.168.110.1 mask=255.255.255.0 mac=cc:8c:bf:8f:dc:1c 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-41,ssid=wifi_marius,bssid=74:83:c2:73:91:90 ,channel=13,cipher_type:
Info:GEN:9 Button_OnLongPressHold
Info:GEN:9 Button_OnLongPressHold
```

mariusbach, 2022-11-01:
I just noticed the new issue, I propose to continue there: #389

closed 2022-11-01 by mariusbach