openshwprojects/OpenBK7231T_App #389: "StartDriver TestPower" - > device dead. There is indeed an issue.

openshwprojects, 2022-11-01:
Our fellow users were right. I will be looking into it right now. Just do "startDriver TestPower" on T and you will get a crash.

valeklubomir, 2022-11-01:
The only sprintf (%f ) is in MQTT_PublishMain_StringFloat

openshwprojects, 2022-11-01:
@valeklubomir i am testing a potential very crude but still working vsnprintf replacement, wait 

but in general, I just did a "testLog" command on this device and also got a crash...
[image]
I know this issue for a very long time already, if you look into newleddriver.c:
[image]
the strange thing is that in the past I had %f issues on N and now they are on T...

but again, maybe I will know more soon, testing.

mariusbach, 2022-11-01:
If it helps nailing it down, the issue appeared upon introduction of .106; with .105 all is fine on my end.

openshwprojects, 2022-11-01:
**I have added a vsnprintf lightweight replacement and no longer have any crashes.** Please wait until tomorrow morning for a commit. I have to do more tests.
The lightweight vsnprintf2 is cool, very little stack usage, thread safe, etc (well maybe it relies on itoa from compiler  but still).

openshwprojects, 2022-11-02:
Some update from me. @iprak  @valeklubomir 

As said yesterday, the "testLog" command crashes my WB3S because of %f in addLogAdv. The "startDriver testPower" also crashes my WB3S, because of HTTP printfs.

Now I have gradually replaced addLogAdv vsnprintf with a light-weight version and testLog command no longer crashes.

"startDriver testPower" was still crashing, so I replaced HTTP printf vsnprintf and now it also works again.

**So using our own lightweight vsnprintf seemingly solves all issues. The only problem now is we will miss some more advanced sprintf syntax....**

openshwprojects, 2022-11-02:
For reference; here is the code I work with:
https://github.com/openshwprojects/OpenBK7231T_App/commit/8b844c7f78a195332365d47d9df91e7e0cb168c7
It seems to fix almost all crashes, all immediate crashes (crash on "testLog" command, crash on "startDriver TestPower", crash on publish receive), but device is still not stable for me.

the code above is not enabled, just included in App, so you can play with it.

NOTE: do not try to just replace original "sprintf" or smth with out sprintf . I tried once and it resulted in WiFI issues. That is why I prefer to use #define to override sprintf instead of a linker method.

iprak, 2022-11-02:
What code change actually led to this issue? That can help us identify how to prevent something like this in future.

openshwprojects, 2022-11-02:
@iprak to be honest, I am not sure. But I know for 100% that I had a "addLogAdv %f" crash really months and months ago. So the printf/sprintf/vsnprintf issue is a long term one. At least that I think may be fixed now, but NOTE: I didn't enable my sprintf etc etv vsprintf yet in the build, just submitted the code.

It is very possible that sprintf/vsnprintf %f crash is just an effect of some other problem, threading or memory, but at least code will be safer now

iprak, 2022-11-02:
I noted that my W600 was not impacted by this, probably because all drivers are disabled for that chip or maybe the printf is just different there? Also, the 7231 where I did encounter the issue, no drivers were enabled.

I am going to put a separate idea for allowing OTA from native webApp so that one could upload firmware locally in AP mode. I was unable to downgrade the firmware and had to open the device and use uart approach . What do you think about this?

An unrelated question: Have you considered starting a Discord server for this project? We tend to use issues/comments for communicating and maybe Discord might be faster in that respect? We can still summarize our discussion into PR so that we don't lose it.

btsimonh, 2022-11-02:
I was also suspecting sprintf a few weeks back - which led to the memtest code...
The vsnprintf function I found came from someone who obviously had problems in lwip - because it had been modified to print IPs directly from an lwip structure.  However, I never got to trust it.
Maybe write a function 'char *s getFloatStr(float f, char *s, int len)' == multiply by 1000, itoa, insert '.', and then search our code for %f and replace with %s and use the fn?
My worry about raw %f is 1/3 = 0.33333333333333333333333333333333333333333333333333333333
but we will never need more than 0.333?

openshwprojects, 2022-11-02:
I have experienced %f addLogAdv issue for the first time many months ago. It was when I created cmd_newLedDriver.c. I have tested it only on T and it worked great with addLogAdv("%f"), but then someone got it crashing on N, and that's how I realized.
And now... %f is crashing for me on T, too. Very strange.

@iprak  it's good you're testing W600

@iprak  I really don't like Discord. It's out of our control, it's not indexable in search engines and it's messy. Elektroda is IMHO way more reliable, with somewhat more than 20 years of history, the biggest forum in Poland and I do my best to promote them to the world. Futhermore, Elektroda provides me support as well, like sending me devices for testing

Still, I see your point - can you send me your Discord on Elektroda Private Message? I think we could benefit from having a private dev-only server. We must do our  best to resolve all issues swiftly.

@btsimonh we can check that, also I have managed to get a script-only build working on windows, so I will check, wait a second

mariusbach, 2022-11-02:
> I was unable to downgrade the firmware and had to open the device and use uart approach

@iprak What worked for me was 5x powercycle to start in AP mode and disable all drivers and MQTT (!). Then the device would find it's way back to the normal wifi network and I was able to do OTA as usual there.

openshwprojects, 2022-11-02:
@mariusbach in my case turning off MQTT server or disabling MQTT really helps a lot and it don't crash any more. 

I pushed a win_main that is able to run SPRINTF replacement and my Scripting stuff on Windows:

https://github.com/openshwprojects/OpenBK7231T_App/commit/0b536684909dbb5f48e2ed563b1f934c1e3fc372

mariusbach, 2022-11-03:
@openshwprojects Thanks, but MQTT is essential for me. Perhaps you could leave a note somewhere (if not done, I haven't checked), that B7231T devices should stay on version .105 for the time being. I use plugs with this chip only for energy monitoring, not much for switching at all.

iprak, 2022-11-03:
> > I was unable to downgrade the firmware and had to open the device and use uart approach
> 
> @iprak What worked for me was 5x powercycle to start in AP mode and disable all drivers and MQTT (!). Then the device would find it's way back to the normal wifi network and I was able to do OTA as usual there.

That has not worked for me. While the device does go into AP mode, it fails to grant ip address. I have tried connecting from multiple phones with same results. I plan to view the serial log before re-flashing it.

btsimonh, 2022-11-04:
float print issues should now be resolved....
@iprak - if you still need to discuss firmware flashing, you could open a new issue?

closed 2022-11-04 by btsimonh

iprak, 2022-11-04:
I tried but was unable to connect to the device in AP mode. I just reflashed it and it right away connected to WiFi. This is definitely not a practical approach which is why more testing and separate build should be used as public releases.