openshwprojects/OpenBK7231T_App #390: [Process] Different release paradigm / more checkpoints

iprak, 2022-11-01:
Currently, there is limited verification being done before producing new releases and every merge seems to generate a new release.

I propose that we produce production releases from a separate branch one which has been used for a while and keep the current main as dev branch which generates beta builds.  The core code impacts numerous devices and it is just not practical for one person to verify all of them.

btsimonh, 2022-11-02:
A good idea.  

Who has the experience to get this in place?

(I just upgraded a device from local build latest to 1.14.109 from github, and it failed to start.  I think the lwip changes are now included in T?  the killing thing was 'startdriver SSDP' in startup line - which worked fine in locally built (assume->old LWIP).  I'm guessing that where previously it failed to get a socket, now it crashes - will investigate).

openshwprojects, 2022-11-02:
@btsimonh  right now I think there is a freqeuent call to sprintf in LWIP  and it manifests in random errors 
[image]
i am on it