openshwprojects/OpenBK7231T_App #391: Feature/help request, Domoticz MQTT support

Jasper2991, 2022-11-01:
My appologies if this is not the correct way of asking for help, my Git-skills and experience are limited.

I managed to Flash OpenBK7231T/OpenBeken to a whitelabel tuya smartplug unit from my local ‘Kruidvat’ and managed to figure out the pinout for the functioning (i would love to add this to the collection of knowledge if this is helpful btw).

Now I wanted to integrate this unit in my Domoticz system and came to the conclusion I do not have the faintest idea how to manage this with OpenBreken. Until now I used Tasmota which I can directly link to my MQTT server, it is basically plug-and-play.  

With the awesome work you all have done I was very hopeful you could point me to a source of info to get me started. I also read a few more comments (like 5) regarding domoticz support, and it may be coming in the future?

All help is welcome, Thanks!

ReanuKeeves01, 2022-11-02:
You should be able to do almost exactly the same thing in OpenBeken, if you go into the config you should see the MQTT option there, select that and configure it from there.  

[image]

Then perhaps use something like MQTT Explorer to check if all is working as it should in regards to the mqtt side. As a Home Assistant user I can't speak of the final integration in Domoticz..

GravityRZ, 2022-11-03:
i tried this as well for an RGB light.
i think the correct way of doing this is

on domoticz enable MQTT autodiscovery
Accept new hardware devices to ON

on OpenBK
configure MQTT
home assistant configuration
start home assistant discovery(this will broadcast the parameters through MQTT)
after that switch the light on/off

the device should be created with dirirectional control

Now with a light which has no PWM, relay or powerdriver running you can not do this because you get this error
[image]
This should be fixed.
workaround is to add a relay to p0 to skip this message.(but then you get an extra light generated)

i however could not get the RGBWW light to work because everything was working except for the colors
dim WORKS
ON DOES NOT WORK
OFF WORKS
selecting colors DOES NOT WORK, Light switches to cold white only

vinibali, 2024-02-23:
My feature request for somewhat the same topic:
https://github.com/openshwprojects/OpenBK7231T_App/issues/1100

vinibali, 2024-09-29:
using the steps here was just fine to setup a general temperature and humidity sensor with CBU board and CHT8305 (Tuya TH01). thanks for the steps!

vinibali, 2024-09-29:
@GravityRZ did you check that with some latest Domoticz and OpenBK version?

GravityRZ, 2024-09-29:
no i did not.
i found that the mqtt implementation on domoticz is not 100% the same as with home assistant so i gave up.