openshwprojects/OpenBK7231T_App #395: Alternate sprintf causeing crash under.

valeklubomir, 2022-11-02:
Both platforms BK7231N and BK7231T cause crash under new sprintf2.
Probably missong some formatting options used in the code.
Example:
[image]

valeklubomir, 2022-11-02:
@openshwprojects 
I hope this was not released.

valeklubomir, 2022-11-02:
I upgraded 5 deviced and all are bricked. 
Forcing SAFE mode. 
After disabling MQTT devices stays active and accesible over network.
But OTA fails. end reboot device.
[image]

Tested on BK7231N and BL7231T

valeklubomir, 2022-11-02:
PLEEEEEEEEEEEASE do not present the new commits as release, at least until 10-20 experienced user/developers can prove that is is working. Mark the as test release.

valeklubomir, 2022-11-02:
OTA fails because snprintf "%u" is not supported.

valeklubomir, 2022-11-02:
More format missing: 
cJSON: "%1.15g", "%1.17g", "u%04x"
drv_itr: %lX %X
rest interface : %.*s
memtest: %08X

openshwprojects, 2022-11-03:
That's strange, I enabled it in one of the latest commit, but I can undo the release.

It was tested on Windows, BL602 and BK7231T.

openshwprojects, 2022-11-03:
One of the other reasons sprintf was hastily pushed:
https://github.com/openshwprojects/OpenBK7231T_App/issues/387

openshwprojects, 2022-11-03:
I have did a clear build and cleared 100% of memory of WB3S (this button with Chinese character on bkWriter 1.60 does that, it's a erase button, it can erase given range of flash, including whole configuration) and flashed with latest version (with sprintf disabled again), I entered WiFi creds, then I loaded this config:
[image]
then type in console: "startDriver testPower" and device is dead within one second.

@valeklubomir , I don't want to make the same mistake with hasty changes, but what do you think? Shall I, for now, just change the vsnprintf used in addLogAdv and httpPrintf255 or whatever it was called?
Does the same crash happen for you?

valeklubomir, 2022-11-03:
@openshwprojects It was just inconvinience for me. For others it is catastrophic. Took me one hour to resolved cables and flash all devices back. Crash occured on all my devices: Plug BL0937, Plug BL0942, Touch panel TuyaMCU, Ceiling list PWM, Light bulb PWM. I was too hasty also, usually I flash first devices which are connected to programmer. And check them. Now it was late and wanted to test new snprintf2, and flashed them all, without waiting for the test devices. Bad decision. 

I will definitely keep one BK7231N and T as test devices in the sandbox, and once version does pass all tests, then upload to other. Actually this could be new feature like auto-diagnostic. Single command executed which will test all critical point which we collected so far.
 
I had to break open one crashed device, and this process can not be repeated very often. It has poor internal design from point of view, when trying repairs, it was constructed like one way device, once broken it is trash.
 
How does work OTA through Javascript Panel in non-safe mode? We could use OTA in safe mode. Can we upload firmware in AP mode/safe mode?

Regarding bk_writer, I used it twice. I rather not do full flash erase. Because also MAC is erased. And then I have to select new, usually do MAC+1 from other device.

valeklubomir, 2022-11-03:
Problem it the variable arguments list in the snprintf2. You need to pull all variables from stack before return from function. When then number of %fnt and number of parameters passed to function do not match then stack after fuction call executes return to wrong call address. If we find % and do not recognise format we still have to pull parameter from stack. May be this way we can filter all undefined formats one by one.

openshwprojects, 2022-11-03:
That's not good, I apologize for that sprintf2 issue, it didn't occur to me and, as said before, it was a hastily decision because it fixes issues in addLogAdv.
**I will say again - current build (idk exactly where it started, maybe LWIP, maybe not) on BK7231T will crash immediatelly (maybe within one second) for me on "startDrivers TestPower" and "testLog" commands, while it's working 100% OK if I just replace addLogAdv asnd htPrintf255 vsnprintf with my version.**

Can you check if the same crash occurs to you or is it specific to my two WB3S devices? Can you check if the crassh occurs if you do a vsnprintf resplacement in those two places I mentioned?

_How does work OTA through Javascript Panel in non-safe mode? We could use OTA in safe mode. Can we upload firmware in AP mode/safe mode?_
@btsimonh  knows more about it, but in safes mode the resst interface is not initializesd. We could add a button in safe mode that says "Init REST functionality" or just add a console command. This is becauses the Safe Mode was designed to run with all modules/features turned off so we can recover device from bad config (eg. remove some pins from Configure Module).

The ideas with removing arguments from stack when found unsupported parm is also very good!

PS: @valeklubomir  special thanks for figuring out the trick with disabling TCP send from MQTT LWIP "publish" and "sub" calls. I think you know what I am talking about. I did that for BL602 and stability of this platform has incresaed.

btsimonh, 2022-11-03:
clear explanation of float vs double in va_args:

https://stackoverflow.com/questions/23836118/variadic-function-va-arg-doesnt-work-with-float-while-printf-does-what-the
https://stackoverflow.com/questions/11270588/variadic-function-va-arg-doesnt-work-with-float

so... in any function using va_args, you MUST extract floats as double ....

openshwprojects, 2022-11-03:
Well @btsimonh , my current proposal is simple - use the vsnprintf replacement for addLogAdv and for htPrintf255 or whatever it was called. I am outside of home now and I don't have access to BK devices, I will be back today evening. 

If you can double-check if my idea would work, please do. I don't want to repeat the mistake with commiting untested solution.

The main question still lingers - why has T platform stability decreased,

valeklubomir, 2022-11-03:
I don't know if this matters here, but may be some idea could be sparked:

- in ESP32 devices I had issues with float. Especialy when running from ISR. Reason was that ESP32 hardware coprocessor can work with float and all operation with float were executed in HW. Coprocessor does not like using in ISR. To avoid this we had to replace all float with double which is handled in SW library only, which could be safely used in threads and ISR. 
- ESP32 also uses serial FLASH as program memory. Once you mentioned that flash in SPI identifies as 25Q16. Operations writing and reading flash in ESP32 could not be execute randomly. And saving to flash was then limited to one thread only. May be all issues were amplified once we started to save data in Driver.
- ESP32 uses Freertos (10.4.3) and LWIP (2.1.2) as BK7231 freertos (9.x) and LWIP (2.1.3)

btsimonh, 2022-11-03:
@valeklubomir - please could you pull and test this branch:
https://github.com/btsimonh/OpenBK7231T_App/tree/testfloats

I have used a fairly well supported nanoprintf from here:
https://github.com/charlesnicholson/nanoprintf

The only down side at the moment is it seems to default %f to 6 decimal places if no precision is given.
i.e. 3.14 prints as 3.140000

If you take a look and think it's a workable approach, tonight I will make __wrap_ functions for vsnprintf, snprintf, sprintf, and propose application.mk changes for T and N.
I don't have any of the other platforms (and have not had any experience of them), so don't know if the __wrap will work there.

(if you are not familiar with the wrap stuff, I wrote it up in the wiki:
https://github.com/openshwprojects/OpenBK7231T_App/wiki/T-&-N-Hardware)

@openshwprojects -
"#define printf addLog"
may have had wide ranging consequences - like
vsnprintf() -> vsnaddLog()
See replacement in new_common.h...  this commit is probably essential to include.
I'm not even sure we have addLog()???

btsimonh, 2022-11-03:
hmm... other platforms do not have inittypes.h?
https://pubs.opengroup.org/onlinepubs/009695399/basedefs/inttypes.h.html

In file included from ../sharedAppContainer/sharedApp/src/new_common.c:17:0:
../sharedAppContainer/sharedApp/src/nanoprintf.h:60:22: fatal error: inttypes.h: No such file or directory
 #include <inttypes.h>

openshwprojects, 2022-11-03:
@btsimonh that is not possible: 
[image]
#define does not expand substrings. It only replaces if string matches 1:1 the given one. Other than that, I will look into it. I don't know about inttypes, tough.

btsimonh, 2022-11-03:
yep, basically that, and have a read of that repo - see if you like it.
I know you found an alternate vnsprintf at some point, I found one, and did not like it (completely unknown source), @openshwprojects found one and it was buggy and supported not a lot.
So your opinion on this one would be good :).
If you think it's likely to have no major flaws for us, I'll do the wrapping so we change EVERY sprintf, vsprintf, snprintf and vsnprintf in the whole codebase, and try on my device which has wired serial :).

btsimonh, 2022-11-03:
> @btsimonh that is not possible:

put it back, and see strange warnings from inclusion of nanoprintf.h (line ~13)
I swear something VERY strange was going on....

openshwprojects, 2022-11-03:
@btsimonh I am not convinced it is possible for #define to act that way.
Please see: https://ideone.com/y3XE9U

btsimonh, 2022-11-03:
> the stack size required and speed is a totally other thing

this one is designed for embedded (see repo linked above) - plus by wrapping and not referencing _real_ functions, we will save some flash :).

let me try here.  If the original functions have an issue, I'd rather nothing used them.

valeklubomir, 2022-11-03:
Runnung TestDriver on T. 
It look like SW deadlock, endless loop. 
After Command Start TestPower is executed, timer executed once "Main_OnEverySecond" completely and then everything is "frozen". I will add Button/LED to simulate Button/Relay, to check status of  QuickRunFrame. 
But MCU does not freeze it has to run always, except deep HW failure. 
CPU has exception to detect failure regarding access to memory. I thing we got it, Because exceptions for Undefined instruction, prefetch abort, data abort end in endless loop:
[image]
Examinig STACK frame could help us to locate program section when it occured.

btsimonh, 2022-11-03:
I just did the wrapping, and my device is running.  TestFloats and TestLog are as expected, no crash.
Ref vectors: I looked at trying to modify our vectors, but was unsuccessful.  I seemed unable to replace them - even played with VTOR.

startdriver testpower
Also survives for me.  (I guess it's simulating something?  I see mqtt traffic...)

will push on the same branch.
You will need:
```
CFLAGS += -DWRAP_PRINTF=1
LFLAGS += -Wl,-wrap,vsnprintf
LFLAGS += -Wl,-wrap,snprintf
LFLAGS += -Wl,-wrap,sprintf
LFLAGS += -Wl,-wrap,vsprintf
```
in application.mk

btsimonh, 2022-11-03:
@valeklubomir - please pull my changes and modify your application.mk, and try again.
I want to hear some positive results! :).
I have a little time tonight.  Wife is out for a few hours, so let me know...

btsimonh, 2022-11-03:
Now running the same thing on N.
Again, all tests run and no death (yet).
(no mqtt logs on N - I guess they are in the OS?)

Had to exclude printf.c in application.mk

mods:
```
SRC_C += ./beken378/driver/uart/uart_bk.c
#SRC_C += ./beken378/driver/uart/printf.c
SRC_C += ./beken378/driver/wdt/wdt.c
```

```
LFLAGS += -Wl,-wrap,malloc -Wl,-wrap,_malloc_r -Wl,-wrap,free -Wl,-wrap,_free_r -Wl,-wrap,zalloc -Wl,-wrap,calloc -Wl,-wrap,realloc  -Wl,-wrap,_realloc_r
#LFLAGS += -Wl,-wrap,printf -Wl,-wrap,vsnprintf -Wl,-wrap,snprintf -Wl,-wrap,sprintf -Wl,-wrap,puts

CFLAGS += -DWRAP_PRINTF=1
LFLAGS += -Wl,-wrap,vsnprintf
LFLAGS += -Wl,-wrap,snprintf
LFLAGS += -Wl,-wrap,sprintf
LFLAGS += -Wl,-wrap,vsprintf
```

valeklubomir, 2022-11-03:
@btsimonh TestDriver on T version is stable.

btsimonh, 2022-11-03:
both T & N also survive IRBlast test (button on remote held down for 2 mins, whilst getting http logs, etc.,) and wifi did not die, and I assume mqtt ok!!!!

btsimonh, 2022-11-03:
@openshwprojects - i have to pop out for 40 mins.  But after I get back I can prepare a cleaner PR, and PRs for the two SDKS?  let me know... 
Good team effort....
br,
Simon

openshwprojects, 2022-11-03:
@btsimonh please do prepare a PR. If there is anything to check for me, also do tell. I was less active today. I will have more time tomorrow.

btsimonh, 2022-11-03:
@valeklubomir - PR is in, sdks updated.
Only remaining question is other platforms, and do they need the wrapping.

valeklubomir, 2022-11-03:
Found difference between T and N regarding printf.
N version has local copy of printf.c  at platforms/bk7231n/bk7231n_os/beken378/driver/uart/printf.c
T version did not have it so GCC used version from LIBC
May be that was the difference.
But now we have new printf nanoprint may be better results.

valeklubomir, 2022-11-04:
I can confirm this version is working on all my N and T device:
- BK7231N + BL0937
- BK7231N + BL0942
- BK7231N + TuyaMCU driver
- BK7231N + LED PWM 
- BK7231N + BL0942
- BK7231T + LED PWM 
- BK7231T + BP5758D
I keep them online to see longterm stability

openshwprojects, 2022-11-04:
@valeklubomir we have a report that led color MQTT topic has now invalid string value, see latest issue

btsimonh, 2022-11-04:
printf.c is now excluded from the N build.
I think we can close for now - maybe raise another issue for porting to W600, W800, etc.?

closed 2022-11-04 by btsimonh