openshwprojects/OpenBK7231T_App #396: MQTT ErrMsg: mqtt_host 192.168.1.50:1883 not found by gethostbyname

GravityRZ, 2022-11-02:
hello,
i am running the latest OpenBK version 1.14.115

the light  is working ok and decided to configure MQTT

MQTT is running inside a docker container and it is currently being used by domoticz, homebridge and a shelly  1PM wifi switch.
they al use the same MQTT server on 192.168.1.50 port 1883

The OpenBK unit(192.168.1.160) however does not seem to recognise the MQTT server
it gives the error

MQTT ErrMsg: mqtt_host 192.168.1.50:1883 not found by gethostbyname

when i start mqtt explorer it is clearly running and all data is coming in(except from the OpenBK unit)

why is it not recognised?

mosquitto v 1.5.7
MQTT v3.1.1 broker

openshwprojects, 2022-11-03:
It seems my latest quick fix for BK7231T stability loss (new sprintf) does not cover some of the formats. Please temporary downgrade the older release. I apologize for inconvenience.

GravityRZ, 2022-11-03:
i am down to version 111 but still not luck

with plain ip config 192.168.1.50 and port 1883 i get
MQTT ErrMsg: mqtt_client_connect connect failed

with http://192.168.1.50 and port 1883 i get
MQTT ErrMsg: mqtt_host http://192.168.1.50 not found by gethostbyname

i am a new user so i do not have a reference of a previous working solution.
what i do know is mqtt on my side seems to be working with all the other equipment

GravityRZ, 2022-11-03:
UNBELIEVABLE
i found the bug
the user and password(which i do not use) can not be left blank

if you fill in 2 spaces or just mosquitto mosquitto MQTT starts working

please fix this irritating bug (costed me 3 hours 2 figure out)

openshwprojects, 2022-11-03:
Oh no! You are correct, it seems.
@valeklubomir 
This is new LWIP:
https://github.com/openshwprojects/OpenBK7231T/blob/master/platforms/bk7231t/bk7231t_os/beken378/func/lwip_intf/lwip-2.1.3/src/apps/mqtt/mqtt.c
This is old LWIP:
https://github.com/openshwprojects/OpenBK7231T/blob/master/platforms/bk7231t/bk7231t_os/beken378/func/lwip_intf/lwip-2.0.2/src/apps/mqtt/mqtt.c
It seems that my quick-fix for empty password/login was not ported to new LWIP!
[image]
Maybe we need to fix it in the App so it won't happen again..

valeklubomir, 2022-11-03:
@openshwprojects Intention of LWIP is to set parameters for username and password to NULL if empty.
We have two options:
LWIP to bypass length check, and allow connection with correct beheaviour.
APP  place NULL when user name and password empty.

openshwprojects, 2022-11-03:
@valeklubomir ok since you think it's save, I will add a fix in next 15 minutes

GravityRZ, 2022-11-05:
tested it with build 128 and it works.
closing this

closed 2022-11-05 by GravityRZ