openshwprojects/OpenBK7231T_App #398: Setting hue via script not working

Handy-E, 2022-11-03:
Tried the following script to simulate a rainbow:
```
restart:
	led_saturation 100
	led_hue 0
	led_dimmer 100
	setChannel 11 $led_hue
	setChannel 12 1
again:
	if $CH11>100 then "setChannel 12 -1"
	if $CH11<0 then "setChannel 12 1"

	addChannel 11 $CH12
	led_hue $CH11

	delay_ms 50
	goto again
```
However, LED stripe is set to "FF0000" (red). I can see CH11 is counting up and down, but hue is not changing.
What am I doing wrong?

openshwprojects, 2022-11-03:
Thank you for testing scripting engine! That's a very good script, I will add it to the examples. I will help you to get it working.

From what I can currently seen, the "$led_hue" reading is not exposed yet. Can you wait a hour or two, I will add this feature for you.

Handy-E, 2022-11-04:
> From what I can currently seen, the "$led_hue" reading is not exposed yet. 

@openshwprojects : Thanks! "$led_hue" seems to be exposed now. But the problem seems to be with
`led_hue $CH11`
where "led_hue" is not set to the value of "$CH11" (which flows between 0 and 100).
"led_hue" seem to always stay at 0 (which is red, 0xFF0000, afaik).
Setting led_hue to a fixed value withing the script, e.g.
`led_hue 80`
works.

Handy-E, 2022-11-05:
So, here's a shorter version of the rainbow-script. It is, imo, more readable and doesn't need a separate channel.
```
restart:
	led_dimmer 100
	led_saturation 100
	led_hue 0
again:
	led_hue $led_hue+1
	if $led_hue>=360 then "led_hue 0"

	delay_ms 50
	goto again
```
Don't know if it works correctly, because as already written above, setting "led_hue" is not working here.
`led_hue $led_hue+1` always results in "led_hue" being zero afterwards.

openshwprojects, 2022-11-06:
Huh, thanks. It was simple that we were using older system for getting arg for hue. Nice catch. It was just an "atof" instead of Tokenizer usage. It should be fixed in next commit in about 15 minutes.

Handy-E, 2022-11-07:
Perfect! It is working now! Thank you very much. And here... the final version, "smoother" and with 360° hue, for a full rainbow experience:
```
start:
	led_dimmer 100
	led_saturation 100
	led_hue 0
again:
	led_hue $led_hue+1
	if $led_hue>=360 then "led_hue 0"

	delay_ms 200
	goto again
```
Feel free to add it to the examples ;-)

closed 2022-11-07 by Handy-E