openshwprojects/OpenBK7231T_App #399: JWBR2S (WBR1)?

B3DTech, 2022-11-03:
Wondering if it's possible to get this working for a JWBR2S with WBR1 module on it.

Thanks.

HumbleDeer, 2022-11-04:
**As per what Tuya notes:**

**JWBR2S**
> JWBR2S is a Wi-Fi module that is suitable for serial communication. It consists of a WiFi +Bluetooth **WBR1** module, and a level conversion circuit.

**WBR1**
> WBR1 is a low-power embedded Wi-Fi and Bluetooth module that Tuya has developed. Embedded with the Wi-Fi network protocol stack and rich library functions, it consists of a highly integrated RF chip **(W701).**

Meaning that the JWBR2S uses the WBR1 as an internal module but **does not expose** any of the general-purpose IOb (GPIO) pins that may be available. If you are looking to use a device that uses this, look elsewhere for the controller that controls the actual device. The JWBR2S module does the same as the old ESP8266 did for Arduino projects - just a wifi dongle.

This probably means that this device is supportable by the OpenBK project but through quite a lot of modifications (simplifications essentially) as it'd have to strip down and communicate over UART solely. It's to be seen/investigated whether custom firmware on these modules would be useful at all, given they may work without modification as long as UART communication protocols can be figured out.

### Tuya Developer docs
[JWBR2S-3.3v](https://developer.tuya.com/en/docs/iot/jwbr2s33v-module-datasheet?id=K9mxzqyzkk6y1) and [JWBR2S-5v](https://developer.tuya.com/en/docs/iot/jwbr2s5v-datasheet?id=K9mxm59oqep1q) (two voltage versions)
[WBR1](https://developer.tuya.com/en/docs/iot/wbr1-module-datasheet?id=K9duisiao4qpa)

divadiow, 2025-02-13:
https://www.elektroda.com/rtvforum/topic4097185.html