openshwprojects/OpenBK7231T_App #400: multiple http commands are not executed

GravityRZ, 2022-11-03:
Hello,
i tried something simple

put the light on 40% brightness warmwhite
http://192.168.1.160/cm?cmnd=led_dimmer%2040%20led_temperature%20500
only first command is executed

put the light on 80% red
http://192.168.1.160/cm?cmnd=led_dimmer%2080;led_basecolor_rgb%20FF0000
only first command is executed

so switching from WW to RGB works as long as you do it with one command

as far as i can tell Tasmota can do this, so multiple commands in one single line

so first command is executed but not second
does not make a difference if you seperate them with either ; or %20

openshwprojects, 2022-11-03:
That's a good find, I think we can support that. But have you tried backlog?

GravityRZ, 2022-11-03:
yes, tried it and seems to be working

maybe put the backlog command under the console commands because i did not notice it on the page

red 80%
http://192.168.1.160/cm?cmnd=backlog%20led_dimmer%2080;led_basecolor_rgb%20FF0000

ww 10%
http://192.168.1.160/cm?cmnd=backlog%20led_dimmer%2010;led_temperature%20500

closed 2022-11-03 by GravityRZ