openshwprojects/OpenBK7231T_App #401: Light on/off state not remembered after reboot or power off/on

GravityRZ, 2022-11-03:
version 1.14.121

flag12 ON (should save power state)
configure startup value -1 (should remember power state)

the RGBWW light seems to remember all of it's setting after power off/on
the powerstate however is always on when restarting or switching off/on

Light Off
restart
Light comes back ON

Light Off
power off/on
Light comes back ON

GravityRZ, 2022-11-05:
to add some more info
this is a driver based light. maybe that is the reason it is not remembering powerstate
[image]

openshwprojects, 2022-11-06:
Should be fixed now?

It was simply not implemented. Saving of the enable was missing.

closed 2022-11-06 by openshwprojects