openshwprojects/OpenBK7231T_App #403: future request maybe use & instead of ; for separation of multiple commands

GravityRZ, 2022-11-03:
i think tasmota and domoticz both use the & sign to combine multiple commands.

i tried to use multiple commands on a selectorswitch in domoticz but it does not stick because everything after the ; is deleted

so instead of 
http://192.168.1.160/cm?cmnd=backlog%20led_dimmer%2045;led_temperature%20500

do
http://192.168.1.160/cm?cmnd=backlog%20led_dimmer%2045&led_temperature%20500

or better get rid of the backlog
http://192.168.1.160/cm?cmnd=led_dimmer%2045&led_temperature%20500

unless there is a specific reason why we are using ; in OpenBK

valeklubomir, 2022-11-03:
No reason not to use both ';' and '&'

GravityRZ, 2022-11-03:
sounds good
let me know when it is implemented and i will test it.

openshwprojects, 2022-11-03:
The experimental scripting feature supports usage of && as a logical operator, so expressions like "setChannel 1 $CH2&&$CH3" can work, so maybe without &&

HumbleDeer, 2022-11-04:
a simple single ampersand (&) may be more in line to what we're used to with things like PHP sites

Handy-E, 2022-11-13:
> so instead of http://192.168.1.160/cm?cmnd=backlog%20led_dimmer%2045;led_temperature%20500

One more question to this example of sending commands: Can anybody tell me how to use "backlog" with MQTT?
E.g., sending "cmnd/dev1/startScript" with payload "myScript" works. But how to send a payload like "backlog led_dimmer 10; led hue 200"? "cmnd/dev1/cmnd" is wrong ;-) Hints, please!
Problem is, that some automation systems like openhab send all data at once, like "270,96,100" (hue,sat,bri). I am able to transform data there, so instead a payload like "backlog led_hue %s; led_saturation %s; ..." is send. "%s" is replaced with corresponding values. But sending 3 MQTT-messages seems not possible without a lot of scripting, so "backlog" could be a saver here.

Also if openbeken itself would be able to send hue,sat,bri,... at once would be a really cool feature.
Maybe, a MQTT-customizer or payload-customizer or something? One could define a "myCustomLEDgetter" like "$led_hue, $led_dimmer, ..." and afterwards you get MQTT-messages "dev1/myCustomLEDgetter/get" with payloads "270, 100, ...".
Proposal for the x-mas whishlist??? :-)