openshwprojects/OpenBK7231T_App #404: mac addres changed

botioni, 2022-11-03:
mac addres chenged after last ota and to C8478C000000
somthing is wrong with thet and cant change mac addres

openshwprojects, 2022-11-03:
are you on BK7231T?
Here is an experimental option:
[image]

do not do this if you are on BK7231N

GravityRZ, 2022-11-03:
i  noticed the same. i am also on BK7231K

the button says dont click so what do we need to do

openshwprojects, 2022-11-03:
It worked for me on BK7231T. 
What is BK7231K? A typo for N? Don't do that for N.
It's for BK7231T, but it was tested on only 1 device so far, you can test if you have flash wires handy in case something goes wrong.
Or wait until a developer other than me also tests it, idk, @btsimonh , @valeklubomir  , @iprak , anyone?

GravityRZ, 2022-11-03:
yes typo
i mean BK7231T

with flash wires you mean open it up and hardwire it?
i came a long way just to avoid that so i will wait to be sure.

iprak, 2022-11-03:
That sounds familiar; I just noticed the same today. The 2 7231T devices which goes hosed report the same mac ending with 8C000000.

I thought maybe this was causing AP connection to fail so after reflashing one I attempted to change its mac but that never succeeded.

openshwprojects, 2022-11-03:
@iprak I don't know why you guys lost MACs today, but in general, there is a partition at certain offset in memory, that begins with TLV header.
If that header is missing, then you are even unable to change MAC, because of bad Beken code.
I have taken a correct TLV bytes (about 512 bytes) and added them to Web App with possibility to flash them to correct offset.
This restores a TLV header for you so you can change MAC as well.
I am looking now into doing the same for N platform.

iprak, 2022-11-03:
I definitely saw serial messages like "TLV header missing" or something like that after I re-flashed the first device.

valeklubomir, 2022-11-03:
I get this "C8478C000000" mac address when flashing with bk_writer and erase entire flash is checked.
So far each flashing with hid_download_py preserved original MAC address and also I never lost it.

openshwprojects, 2022-11-03:
Don't use that option from Web App yet. I suspect there might be some issue with using the RF partition bytes I posted. Do it only if you are dev and have flash wires connected.

iprak, 2022-11-03:
My device's mac ids were not this prior to the bad update. I have ip addresses assigned by mac for all devices and non-unique mac might have been causing connection instability. 

Should 2 people ended up with same mac?

mariusbach, 2022-11-03:
I've got one device which after the BK7231T issues does only work in safe mode and I can't get it out of there. I'll flash it hardwired on the weekend though.

Anyway, the MAC address it tells me in safe (AP-) mode is 02X:02X:02X:02X:02X:02X. Which is weird and may fit this issue.

openshwprojects, 2022-11-03:
@iprak  MAC are supposed to be unique, but when a TLV  header is removed, the Beken is using a default MAC, which is the same for all people.
I am investigating. It seems that even in the latest build, the device stability and HTTP processing has **decreased after restoring the RF partition (taken by another device).**
It seems like RF partition contains device-specific information that can cause HTTP issues.
What is more interesting, is that my device worked better with RF partition removed.
Maybe it would be possible to restore RF partition, but only the section where MAC is stored, and ignore other CAL data.
One would need to look into Beken code how exactly it's stored and where.
Look for: "No TLV Header found in flash".
https://github.com/openshwprojects/OpenBK7231T/blob/eaf4e1a606571b9f13d9c933da61670c712a6c8d/platforms/bk7231t/bk7231t_os/beken378/func/bk7011_cal/manual_cal_bk7231.c
Our MAC address culprit is manual_cal_get_macaddr_from_flash. But there is also some other data which might cause instability.

iprak, 2022-11-03:
Yes I can give it a try.. in a while though.

openshwprojects, 2022-11-03:
**I think I've found the issue. It was most likely a false alarm. My MAC address was not changed in time and having two devices with same MAC on network was way too much too handle for Beken and router and this is why the free memory started decreasing.**

So I think it might be safe to use the RF Restore button now, as long as you make sure you don't have mac duplicates.

openshwprojects, 2022-11-03:
**The button shall now work for both T and N, and also should randomize the 3 bytes of MAC to avoid duplicates.** Still, needs testing to be sure.

GravityRZ, 2022-11-04:
any news on this. i am on V121 at that version states text about the restore rf config

please note thatr with my version it looks completely whiped, not only the last 3 bytes

[image]

iprak, 2022-11-04:
I used that .. successfully.

GravityRZ, 2022-11-05:
also tested it
pressed the button, answered yes and then nothing seems to happen
after 5 minutes i rebooted the device and it had a new mac address
changed the address by changing the last digits and it sticks

seems this solves the issue
hopefully something which breakes this functionality will not break in the future

botioni, 2022-11-05:
Error:CFG:flash_vars_valid - not our magic, erase this is what is erasing rf cfg out warning out thet if you flash ota you wil get mac erased rf config erased

iprak, 2022-11-05:
I think anyone who would upgrade from an older firmware will hit this issue. I ran into it for my 3rd BK7231T. I wonder if this issue surfaces if one has multiple BK7231T devices following a similar upgrade path.

botioni, 2022-11-05:
Anybody how wil flash higher then x.x.106 wil run into this problem for sure it is realocated memory 
Problem if you update or if you do new dosent mather

btsimonh, 2022-11-05:
?? it should not affect the config area?

botioni, 2022-11-05:
Rf restor working you need to restart to see efect

botioni, 2022-11-05:
It can be the flashing effects config area as well not the rf restore ..

btsimonh, 2022-11-05:
Are you running LFS?
the LFS flash is limited to the OTA area, it can't affect the config or RF area.... unless there is an error in my commits?
And it's not active unless you put a file there...

botioni, 2022-11-05:
not runing LFS on mine Bulbs first time discoverd on a new device 0 config on it

GravityRZ, 2022-11-09:
it seems that if we use start save OTA this can happen.

if you are not using the LFS(no scripts) then use the start Quick OTA

bonuzzz, 2022-11-24:
I lost mac addresses on two devices today. The bulb upgraded from .50 and the plug upgraded from .38. Both are 7231T
So, is it possible to rewrite mac addresses on these devices?

openshwprojects, 2023-02-22:
The restore option should be already well tested, so I can close this issue.

closed 2023-02-22 by openshwprojects